iNavFlight/inav #4349: 2.1RC3 runtime calibration fails to complete on SPRF3

stronnag, 2019-02-11:
Prior to d2c60a60e524c6742bd26643ea9cffa32f0fad08, runtime calibration always works (e.g. 5acd8f099 works perfectly).

Post d2c60a60e524c6742bd26643ea9cffa32f0fad08, run time calibration fails to complete (at least in a timely fashion), arming is not possible.

```
# version
# INAV/SPRACINGF3 2.1.0 Feb 11 2019 / 15:58:41 (db874aa42)
# GCC-8.2.0

# status
System Uptime: 327 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=72MHz, GYRO=MPU6050, ACC=MPU6050, BARO=BMP280
STM32 system clocks:
  SYSCLK = 72 MHz
  HCLK   = 72 MHz
  PCLK1  = 36 MHz
  PCLK2  = 72 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10002000, Heap available: 488
I2C Errors: 0, config size: 3769, max available config: 6144
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 16, cycle time: 1008, PID rate: 992, RX rate: 49, System rate: 9
Arming disabled flags: 0x140a00
```
Having just typed the above, I get three beeps! Getting status shows:
```
System Uptime: 363 seconds
...
Arming disabled flags: 0x140800
```
Having to wait six minutes for something that used to take six seconds is a bit of  a regression. In the field I timed out and reloaded the older firmware ...

CLI `diff` : 
[sprf3-diff.txt](https://github.com/iNavFlight/inav/files/2851959/sprf3-diff.txt)

stronnag, 2019-02-11:
Time to calibrate seems to vary a lot
```
# status
System Uptime: 1502 seconds
...
Arming disabled flags: 0x140200
```
waiting ... at least the GPS has acquired  an indoor fix.

giacomo892, 2019-02-11:
Same applies to the accelerometer calibration. It is very picky with 2.1RC3, very hard to complete all the steps when holding the board with hands.

stronnag, 2019-02-11:
Patience is (evidently) a virtue 
```
System Uptime: 2287 seconds
...
Arming disabled flags: 0x140800
```

stronnag, 2019-02-11:
Maybe a F3 problem; my F4 boards calibrate quickly.

digitalentity, 2019-02-11:
Hm... Maybe I need to find an F3 board.

Regarding accelerometer calibration - it's now more picky when the board is moving during calibration. We can probably relax the calibration threshold a bit

digitalentity, 2019-02-11:
Unfortunately I don't have an SPRACINGF3 board. @stronnag can you enable debug trace and capture debug messages? It should give us a hint what part of calibration gets stuck/

stronnag, 2019-02-11:
On the case.

stronnag, 2019-02-11:
After c. 3 minutes
```
[     0.075] INAV/SPRACINGF3 2.1.0 Feb 11 2019 / 20:02:19 (db874aa42)
[     0.075] Memory allocated. Free memory = 888
[     0.075] Memory allocated. Free memory = 788
[     0.076] Memory allocated. Free memory = 688
[     0.730] GYRO CONFIG { 0, 1000 } -> { 0, 1000}; regs 0x00, 0x07
[     5.539] Gyro calibration complete (-22, 6, -10)
[     7.539] Gravity calibration complete (923)
```

digitalentity, 2019-02-11:
So it's baro getting stuck. Makes sense. Will relax the tolerance

stronnag, 2019-02-11:
Confirmed, 2nd attempt:
```
[DEBUG] [     0.075] INAV/SPRACINGF3 2.1.0 Feb 11 2019 / 20:02:19 (db874aa42)
[DEBUG] [     0.075] Memory allocated. Free memory = 888
[DEBUG] [     0.075] Memory allocated. Free memory = 788
[DEBUG] [     0.076] Memory allocated. Free memory = 688
[DEBUG] [     0.730] GYRO CONFIG { 0, 1000 } -> { 0, 1000}; regs 0x00, 0x07
Unhandled MSP frame 182 with payload [2]
[DEBUG] [     5.539] Gyro calibration complete (-23, 5, -10)
[DEBUG] [     7.539] Gravity calibration complete (916)
[DEBUG] [    95.745] Barometer calibration complete (-15503)
```

digitalentity, 2019-02-11:
@stronnag please test #4351, you can also play with `baro_cal_tolerance` CLI option ([centimeters@sea level] of acceptable deviation during calibration)

stronnag, 2019-02-11:
flashing now

stronnag, 2019-02-11:
75 seems to work, but ...
```
baro_cal_tolerance = 75
Allowed range: 0 - 0
# set baro_cal_tolerance = 100
Invalid value. Allowed range: 0 - 0
```

digitalentity, 2019-02-11:
Crap, fixing

stronnag, 2019-02-11:
with 75
```
[    49.721] Barometer calibration complete (-15630)
```

digitalentity, 2019-02-11:
Still too long I guess. How noisy is that baro..? We need a better default. 150?

stronnag, 2019-02-11:
with 100
```
[    39.702] Barometer calibration complete (-15621)
```
It's always been a noisy baro.

stronnag, 2019-02-11:
150
```
[    17.604] Barometer calibration complete (-15478)
```
more like expectations, time wise, I suppose.

digitalentity, 2019-02-11:
Yep, let's make a default 150. Even if zero calibration is a bit off, position estimation will zero it out with pre-arm altitude zeroing.

digitalentity, 2019-02-11:
Fixed by #4351, closing

PR #4351 (2019-02-12): Relax baro calibration threshold and make it a configurable option
Fixes #4349
closed 2019-02-11 by digitalentity