iNavFlight/inav #4350: 2.1RC3 Delayed receiver signal with Matek F405 Wing and FPort (Softserial1)

Xplorer88, 2019-02-11:
## Current Behavior
Since I have flashed the latest 2.1 RC(x) versions the receiver isn't working anymore.
I'm using a Matek F405 Wing with FrSky R9 mini receiver running on FPort (Softserial1).
The signals from receiver are delayed or even lost. If I move the stick the wing is reacting 2 - 3 seconds later. It is also visible on the receiver tab in the configurator.
If i downgrade to 2.0.1 everything is working as expected.

Thanks for your help.

## Steps to Reproduce
1. open the configurator and navigate to the "Receiver" menu
2. move the sticks on transmitter
3. delay is visible on bars (or wing if battery is plugged in)

## Additional context
https://pastebin.com/kJKv57JP

---

# version
# INAV/MATEKF405SE 2.1.0 Feb 10 2019 / 16:44:59 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

giacomo892, 2019-02-11:
1. Running a receiver on softserial is totally wrong. Please use an hardware UART.
2. Softserial ports are not usable at that speed (115200)

Xplorer88, 2019-02-11:
Hi giacomo892,

thanks for your help, I understand. But why is it totally wrong? Mateksys also describe it on their site http://www.mateksys.com/?p=4488

giacomo892, 2019-02-11:
FPORT --> UART
S. Port --> Can be used on SS

Xedos9er, 2019-02-14:
Also curious about this statement. I got massive problems with failsafes running F.Port on a R9MM with F405 wing using Softserial. Could this be my issue?

teckel12, 2019-02-14:
@Xedos9er You shouldn't connect your receiver to soft serial, the speed is too fast for soft serial.  The reason is that soft serial is just that, it's not an actual UART.  There's software that's running to emulate a serial port on a couple pins.  Problem is, the flight controller needs to do more important things, like calculate stabilization.  So, the soft serial ports can't run at the communication speed that's required by the receiver.

In any case, don't use soft serial for your receiver connection.  It can be used for GPS or telemetry.

Xedos9er, 2019-02-15:
I deactivated now softserial on my f405 wing and connected the R9MM to same pin (TX2) but using the "inverted s.port" out like described by Matek. When deactivating softserial on f405wing, tx2 should run as a hardware UART.
Will do some testing and hopefully my problem with f.port failsafes is gone.

It should be better documented that f.port should never be used on softserial.

giacomo892, 2019-02-15:
Perfect. Closing

closed 2019-02-15 by giacomo892