iNavFlight/inav #4353: "Formation Helper" mode

SimplyKot, 2019-02-12:
## Current Behavior
There is no any "Formation Helper" (FH) mode.

## Desired Behavior
Display on the OSD screen of the approximate position of other UAVs from the formation.

## Suggested Solution
The UAV controllers exchange packets containing a unique identifier, GPS coordinates and GPS height via UART using LORA (if only two UAV are in formation) or mesh network (for tree or more UAV in formainon) 
On the OSD, the approximate direction to the specified point in space is displayed.

## Who does this impact? Who is this for?
All users flying formation or together

Pairan, 2019-02-12:
Have a look ...
#3626 
#4112

SimplyKot, 2019-02-12:
Exactly what I need!
Thank you.

closed 2019-02-12 by SimplyKot