iNavFlight/inav #4358: Inav crashes

rcjarvis, 2019-02-12:
HI yall, been a fpv pilot since the Wii era, hit a lill bump in the road... I really loved Inav thus far. Wanted to try your Tricopter features..but...:
## Current Behavior: When I start Inav, the logo appears, the gui flies to the quicklaunch and the gui dissappears... I see the task bar Icon but cannot acces the gui... I booted and rebooted now I want to try uninstalling/reinstall but I see no Inav in the add/remove soft window.

[image]

## Steps to Reproduce

1.Launch Inav.. dispair...
2.
3.
4.

## Suggested solution(s)
 I WISH TO UNINSTALL THEN REINSTALL INAV BUT I CAN NOT FIND UNINSTALL PROG

## Additional context
I installed Inav this morning and today was the first time I was working with it, I did manage to program Matek f405 std boards with gps/buzzer/led etc... But when it came to my f722 se , I updated FW successfully, but then when I tried to Connect to begin programming, I nav ditched... I then installed FW in Betaflight, again success, but then same bug: unable to connect...

Zadig sees the board and everything worked fine previously so I don't want to mingle with the drivers since I have 2 other drones to finish tuning...

Thx in advance for the time and effort !

Dan
---

rcjarvis, 2019-02-12:
---UPDATE---I installed on another computer and it works fine... could not connect to my F722 se... I doubt it has a bug...all other drones connected fine though...

DzikuVx, 2019-02-12:
https://www.youtube.com/watch?v=XMoULyiFDp4

closed 2019-02-12 by DzikuVx

rcjarvis, 2019-02-13:
Hey PAwel wow, thank you very much, I wonder how come I did not find this video... I made so many searches with keywords "Inav, crahes on launch or Inav taskbar...

I will try this right away !

Daniel Gervais

https://www.youtube.com/watch?v=XMoULyiFDp4