iNavFlight/inav #4361: MATEKF722_HEXSERVO target missing from RC/Final builds

Frontier314, 2019-02-13:
## Current Behavior
The MATEKF722_HEXSERVO target is missing from the Final/RC INAV builds.
This target is needed when using the F722 on a hexa configuration in order to allow servo on S7.

## Steps to Reproduce
make TARGET=MATEKF722_HEXSERVO fails with unknown target.

## Expected behavior
To be able to compile with TARGET=MATEKF722_HEXSERVO

## Suggested solution(s)
Include the target configuration files from the development tree.

shellixyz, 2019-02-14:
The development branch is merged into the master branch only at the time of the final release. If you want to build 2.1.0-RC3 you need to checkout the 2.1.0-RC3 git tag or if you want to build the latest code switch to the development branch.

closed 2019-02-14 by shellixyz

Frontier314, 2019-02-14:
> 
> 
> The development branch is merged into the master branch only at the time of the final release. If you want to build 2.1.0-RC3 you need to checkout the 2.1.0-RC3 git tag or if you want to build the latest code switch to the development branch.

When I check out the 2.1.0-RC3 branch, the MATEKF722_HEXSERVO target is missing, therefor I am unable to build it to test it on my hexacopter.

shellixyz, 2019-02-14:
I added it to the release you can just flash it from the configurator now