iNavFlight/inav #4362: İNAV SERVO MİXER- HELP

xmrtg70, 2019-02-13:
hello. I have a DYS F4 PRO V2 control card. S5 and S6 have connected to servo SG90 but can not control. how do I make the settings. How should servo mixer settings be? kumandadan atamak istediğim kanal 10.
[image]
[image]
[image]

DzikuVx, 2019-02-13:
https://www.youtube.com/watch?v=0cLFu-5syi0

closed 2019-02-13 by DzikuVx

xmrtg70, 2019-02-17:
these settings are not available for the quadcopter. I could not