iNavFlight/inav #4363: Channel 10 not working

Richrfl, 2019-02-13:
## Current Behavior

iNav version 2.1.2
Controller Matek F722-Wing
Transmitter Taranis X9D
Receivers tested FrSky X8R and X6R
Using channel 10 as Mode selection.
Copied channel 10 setting in the transmitter to channels 6,7,8, 11 and 12. All of them follow the switch.
Tested iNav version 2.0.0 with SPRacing F4. Channel 10 works fine.
Verified serial output with a channel decoder and shows channel 10 works.
## Steps to Reproduce

1. Assign channel 10 to a switch in the transmitter
2. Connect receiver to the FC.
3. Go in the configurator to the Receiver tab
4. Check operation of the channel 10.

## Expected behavior

In the Receiver tab I should be able to see the Channel 10 operation.
## Suggested solution(s)

I guess there is some line of code missing to address the channel 10 input
## Additional context

https://pastebin.com/u/Richfl
---

# version
# INAV/MATEKF722SE 2.1.0 Feb 10 2019 / 16:42:36 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

- FC Board name and vendor:
Matek F722-wing
- INAV version string:
2019-02-13 @ 09:51:15 -- Running - OS: Windows, Chrome: 72.0.3626.96, Configurator: 2.1.2

shellixyz, 2019-02-14:
I can't reproduce your issue. Matek F722, X8R, SBus connected to RX2, iNav 2.1.0RC3. Channel 10 and all the other ones work fine

Richrfl, 2019-02-14:
I know it may look nonsense, but did you used the MATEKF722SE.HEX version?. It doesn't make me sense that while the channel is working out from the receiver/converter, the board does not see it. I discard hw failure. I may have something wrong in the FC config, but I haven't see anywhere where I assign control channels to the SBUS. Any idea?

shellixyz, 2019-02-14:
I don't have a Matek F722-Wing only a Matek F722-STD but it shouldn't make any difference. The S.Bus handling code is the same in both cases. I don't think there is anything wrong with your config. Are the channels above channel 10 working ?

Richrfl, 2019-02-14:
Answering your question, yes channels 11 and 12 are working fine. (I have 12 channels enabled on the TX). On the other hand, and for that reason and based on personal experience I wouldn't be surprised that when the SE was generated a line of code was accidentally dropped. Is the only reasonable thing I can think about given that everything else looks OK.

shellixyz, 2019-02-14:
> I wouldn't be surprised that when the SE was generated a line of code was accidentally dropped

Unless it is due to a compiler bug that's not possible. All targets are built from the same code and there is no preprocessor directive that would affect only one channel.

Richrfl, 2019-02-14:
I fully agree with you, I would say though, it's almost impossible. Now, I may have overlooked some diagnostic test, therefore, what I will do (and hopefully it will be fine), I'll store the dump file to keep the configuration, flash the MATEKF722 (same you used) into the F722-WING, and see what happens. I'll not use it for flying but only for reverse test. Do you think I may mess up the board if I flash that fw?

shellixyz, 2019-02-15:
You should avoid doing that. There is a small chance it could hurt the board and it probably won't boot up because the IMU won't be detected.

Richrfl, 2019-02-15:
Thanks a lot Michel. Better be safe than sorry. I am glad I did the comment. On my side, until someone else can check with the same fw and board, the matter is pending from resolution, unless I get another clue that point out that I have something wrong or missing on my side. In the meantime, given that all other channels are working I will change my Mode switch to another channel. (May be when I do that I happen to find something wrong on my setting, although the test with the channel decoder would make that to be almost impossible).
Thank you very much for your attention. If I find something new on this issue, I will comment here. I'll change the channel this morning, therefore we will know soon enough.

shellixyz, 2019-02-15:
I don't want to insult your intelligence but did you check in the channel monitor on your radio that the channel is actually moving ?

Richrfl, 2019-02-15:
LOL Michel. At this point, no question would be insulting my intelligence, even if you ask me if I have a battery in the transmitter.
Yes, I checked the Channel and the Output monitor. 
Anyway, when this morning I kept working in this issue, after verifying again that the other channels were working it seems that something went wrong, to the point that I think the whole SBUS, or the TX itself is faulty. I will have to verify the operation of the TX and SBUS in planes that I am flying using the HobbyEagle A3S2 FC.
Let's drop this ball until I figure out what is going on now.

Richrfl, 2019-02-18:
Michel, I checked the Tx/Rx again and SBUS is OK. I think that I did something wrong when wiring it up and that's what had the SBUS apparently not working. Those are the good news. The bad news, is that still the Channel 10 does not show. In the Taranis, I copied the Channel 10 to Channel 8, and I will use it as a walk around. To be clear, when before I tested channels 7,8, 11 and 12, I just copied Channel 10 into those channels, so that If there was anything wrong in the configuration of Channel 10, no other channel would work. Opposite test, I used the A3S3 FC from Hobby Eagle, using Channel 10 as Mode Control, and it is working just fine.
In summary, in my check list, SBUS channel 10 in iNav MATEKF722SE, is not working. We would need someone using this fw to confirm operation.

shellixyz, 2019-02-28:
I've been able to test with a F722-Wing, X8R and the settings from your dump. Channel 10 is working fine for me.

Richrfl, 2019-03-01:
Thanks a lot for the effort, Michel. It's now obvious I have a problem on my side. I will start by deleting Ch10 mixer and re-editing it from scratch.  If that doesn't work, I will have to ask you some detailed information about your setting both in the TX and the MATEK, but I would not be surprised if my CH10 is messed up some strange way.
Will report back as soon as I can go through this. I am currently trying to walk around the PIFF in flight adjustments based in an airspeed sensor connected to the SPort. 
Would like to ask you a question on a different part of iNav. I was assuming that when you use Manual mode, stick controls are not modified by the FC. However, I noticed that my throws are corrected in the bench. Question is Manual mode is or is not "path through"? 
Thank you

shellixyz, 2019-03-01:
Manual mode applies the manual expo and rate settings then feeds then directly to the mixer. If you have `manual_rc_expo` and `manual_rc_yaw_expo` set to 0 and `manual_roll_rate`, `manual_pitch_rate` and `manual_yaw_rate` set to 100% then manual mode is passing through unmodified channels.

Richrfl, 2019-03-04:
Michel, I finally found out the reason for CH 10 not working and I am reporting just in case some other dummy like gets into the same situation. In a rudimentary walk around to manage the speed/PID (for the Hobby Eagle FC), I forced CH 10 to 0 when the speed was under a certain speed, thus entering into stabilized mode. Over that speed the FC would be controlled by the Mode Switch (CH 10).
Sorry for having you entertained with this.
On the other hand, I set the several manual that you suggested, but it didn't help. The only way I found was to set the Mixer to RC roll, instead of Stabilized Roll, but then it seems that I lack stabilization. I will have to do some more homework about this. I guess that now I should use the support link to follow up at the RC groups site.
Thank you for your help. Hope someone will take some time to implement PIFF gain control Airspeed based soon. Like some other fixed wing requirements. Have a good one

shellixyz, 2019-03-04:
Good, at least now we know :smile:. For the manual mode issue please open a new issue with the details or join the Telegram group. We are waiting to solve the AHRS issue to improve the wind estimation to be able to estimate airspeed and add the speed scaling. But we could add speed scaling based on airspeed read from a pitot. Thing is it doesn't look like a lot of people use pitot tubes it is often unpractical.

closed 2019-03-04 by shellixyz