iNavFlight/inav #4364: MATEK F405-STD - Receiver Mode IBUS doesn't work

pincorr, 2019-02-13:
Hi,
selecting IBUS as recevier mode, all colored bars in Receiver tab are greyed out. 
Selecting SBUS they turn colored.
Tested with all INAV releases from 1.9.3 up to last release 2.1.3 but error is still present.
Any advice to make IBUS option to work?
Thank you

shellixyz, 2019-02-13:
How is your receiver connected and which FC are you using ? Please also provide a config dump.

hali9, 2019-02-14:
It looks like your receiver is giving SBUS at the output.
I do not know what transmitter you have, but in FS-i6 you can change it in the menu.

pincorr, 2019-02-14:
hi all,
here more details:
- f.c. Matek F405-STD;
- receiver flysky FS-IA6B;
- Receiver mode : Serial-based receiver (Speksat,......) - IBUS
- connection of recevier : G to G, 5V to 4V5, IBUS/SUMD to RX2 ( as described in MATEK website at Receivers page).

I enclose dump file

[INAV_cli_20190214_093042_Matek_f405_STD.txt](https://github.com/iNavFlight/inav/files/2864007/INAV_cli_20190214_093042_Matek_f405_STD.txt)

shellixyz, 2019-02-14:
I can't reproduce your issue. Same receiver, same FC, IBus connected to RX2, iNav 2.1.0RC3. Either your receiver isn't outputting what you expect it to or your wiring isn't right. If you got a FS-i6X remote control your can switch the output between IBus and SBus. Maybe it is set to output SBus.

pincorr, 2019-02-14:
Are you saying that IBUS mode works for you ?
I tried with and without receiver connected and nothing works. Also checked IBUS setting on radio control and it's set IBUS. 
If it is was a receiver problem, probably another receiver should work...i suppose. I'll try with another receiver, but i wonder why the bars are grey out with and without receiver connected. Seems more a software bug, or not ?

shellixyz, 2019-02-14:
Yes it is working fine for me. The bars are greyed out because the FC doesn't receive any data. What's displayed in this will be different depending on the selected RX protocol. I used exactly the same hardware and it is working for me so I seriously doubt it is a software bug.

pincorr, 2019-02-14:
At this point I have no idea what to do. What do you suggest as further check ?

shellixyz, 2019-02-14:
I would check:

* With another receiver
* On another FC
* If you can the output of the receiver with an oscilloscope and/or logic analyzer
* You can also add debug statements in the code to see what's wrong if the output of the receiver is fine and you're convinced it is a software issue

hali9, 2019-02-14:
You have enabled telemetry but don't eneble it on uart port
`serial 1 64 115200 38400 0 115200`
Try disable temetry if it not used.

shellixyz, 2019-02-14:
@hali9 Telemetry is not enabled on UART2, disabling the telemetry feature won't help

hali9, 2019-02-14:
Maybe inverter is enabled ?

> Built in inverter for SBUS input (UART2-RX)

pincorr, 2019-02-14:
I will start with another receiver first. I will let you know.

shellixyz, 2019-02-14:
@hali9 There is no inverter on the RX2 pad there is a dedicated pad for that (Sbus) and it is a F405 MCU without built-in UART inverters

pincorr, 2019-02-14:
HI all,

solved ! 

Tried another fS-IA6b receiver and IBUS setting is now correctly working.

Shellixyz and hali9 thank you for your support

shellixyz, 2019-02-14:
Nice !

closed 2019-02-14 by shellixyz