iNavFlight/inav #4365: Resouce Remapping?

anil1112, 2019-02-14:
Hello, I need to change my motor resources.
Because some pin is broken, and I need to connect other pin.
my motor 2 pin : B01 and is broken. 
I want to connect this motor to A01. 
How can I do this?

[image]

DzikuVx, 2019-02-14:
No resource remapping see #1145

closed 2019-02-14 by DzikuVx

anil1112, 2019-02-14:
> No resource remapping see #1145

why is this not possible inav? I can change resource remapping with betaflight.

flythebluesky, 2020-04-18:
I just built a 7" specifically for INAV, and now I find out it is not possible to remap motors? So your FC and ESC's must always face forward???

digitalentity, 2020-04-18:
@flythebluesky look into board alignment and motor mixer. You can place your board in any way you want and you can configure your motor mixer to match the ESC connections.

digitalentity, 2020-04-18:
I'm locking down this issue to avoid noise. Community support is available via https://t.me/INAVFlight