iNavFlight/inav #4370: SURFACE baro fallback ignore speed of 50cm/s

SaschaIoT, 2019-02-14:
## Current Behavior
Use SURFACE mode with VL53L0X rangefinder. In case of bad light, rangefinder got bad readings and baro is used instead to get altitude. If baro is used SURFACE speed of 50cm/s is ignored and copter give full throttle to go to desired altitude.

## Steps to Reproduce
1. Activate SURFACE mode and use VL53L0X with bad light. Ensure VL53L0X got bad readings.
2. Arm copter and leave stick directly in mid throttle position.
3. Copter give full throttle to reach desired altitude.

## Expected behavior
With baro also speed of 50cm/s is used to reach altitude.

## Additional context
[diff all](https://pastebin.com/EvugGJD0)

---

- FC Board name and vendor: [Graupner OmniNXT F7](https://www.graupner.com/Flight-Control-OmniNXT-F7-mit-OSD-BF/48377.NXT/)
- INAV version string: INAV/OMNIBUSF7NXT 2.0.1 (but used with [Fix non-working AGL estimation](https://github.com/iNavFlight/inav/pull/4196) bug fix)

digitalentity, 2019-02-15:
Please use the latest RC of 2.1.0 and test again, please also record BB log if possible.

SaschaIoT, 2019-02-15:
I wrote with @digitalentity. He wrote me that the VL53L0X reads low altitudes on ground around ~50cm in bad light conditions. Copter thinks before arming he is at ~50cm. After arming throttle stick is directly at middle position (copter go up). Rangefinder now read the real ~10cm. Copter thinks he is falling very fast an give full throttle to reach the real ~50cm. This is only a theory, @digitalentity verifying it and close than the bug. In this case the bug fix is to use a better rangefinder (iNav code is correct).

SaschaIoT, 2019-02-15:
Maybe as a workaround we can limit not only the falling speed to 50cm/s. If there was a CLI variable to limit the rising speed for SURFACE mode, bad rangefinder readings has only the effect of copter go slowly down or up. But if rising speed is to slow there could be a problem if the copter has really to go up higher very fast.

SaschaIoT, 2019-03-30:
Solution is to use better sensor. iNav could not detect if sensor readings are good or bad. So there is nothing how software can improve it.

closed 2019-03-30 by SaschaIoT