iNavFlight/inav #4372: IBUS reciever issue

kaliblazin707, 2019-02-15:
Ok, so my reciever tab on Inav reads all 14 channels as 0. I have the reciever wired up correctly, confirmed in betaflight. It shows all 0 even when no reciver is connected, and only when IBUS is saved. I have 3 FCs I've tried this on: Matek F405. Matek F411 Wing. Omnibus F4 Pro. Reciever works in betaflight on all 3. 

I've tried; 
Reflashing every 2.x firmware available. 

Using inav configurator 2.1.3,  2.1.2. Google app 2.1.2 configurator. 

I reinstalled windows, formatting my hard drive, redownloaded the latest inav 2.1.3 configurator, unzipped straight to desktop, reflashed inav 2.0.1, and still, ibus reads all 0s on the reciever tab, on all 3 flight controllers, with or without reciever connected.

shellixyz, 2019-02-15:
It is normal channels read 0 when the FC doesn't recognize the IBus data. What receiver are you using ? Please post a config dump.

kaliblazin707, 2019-02-15:
Really? It ususally says 1500 for me when no receiver is plugged in. Its an x6b, again, that works in betaflight according to the configurator when flashed with betaflight.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/MATEKF411 2.0.1 Oct 31 2018 / 19:29:08 (a04e68aaa)

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature TELEMETRY
feature CURRENT_METER
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 839 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

kaliblazin707, 2019-02-15:
https://imgur.com/a/oQDnMh5

kaliblazin707, 2019-02-15:
https://imgur.com/a/G5N8Ad5
Here's what it looks like when saved as SBUS. this is how the receiver tab should look.

kaliblazin707, 2019-02-15:
Heres a dump with it saved as IBUS

# dump

# version
# INAV/MATEKF411 2.0.1 Oct 31 2018 / 19:29:08 (a04e68aaa)

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature TELEMETRY
feature CURRENT_METER
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
[... 837 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

shellixyz, 2019-02-15:
Yes it is normal. We're going to make it consistent again in the next version. For now (2.1.0) no data received for CRSF and S.Bus = channels to 1500 and 0 for all the other protocols.

I will check an X6B with one of the FCs you mentioned later today.

kaliblazin707, 2019-02-15:
When I plug the reciever in using ibus, setup on uart1 rx1, I still get all 0s. Port tab configured.

shellixyz, 2019-02-15:
This is really strange. I am using IBus on some of my builds from an X6B, never had issues. I also tested yesterday IBus output from an IA6b receiver connected onto a Matek F405 with 2.1.0-RC3 and it is working fine.

kaliblazin707, 2019-02-15:
Yea it works in betaflight. Its only in inav that I have the issue. I bought the f411 for a new nano talon build, and i only use ibus. I was really confused when I seen all 0s on the receiver tab, usually its 1500. I dont understand what I'm doing wrong

kaliblazin707, 2019-02-15:
Hmm ive just noticed nav launch , auto tune and auto trim are missing from the modes tab as well..

shellixyz, 2019-02-15:
Autotune, launch and autotrim flight modes will only show up if you select airplane as platform in the mixer tab

kaliblazin707, 2019-02-15:
Here is another dump with as many settings as i can save with the issue i am having. I put in a new x6b pulled from my drone. still all 0s
# dump

# version
# INAV/MATEKF411 2.1.0 Feb 10 2019 / 16:46:25 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature SOFTSERIAL
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 917 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

kaliblazin707, 2019-02-15:
https://imgur.com/a/KrQNZuV
Here is a picture of my wiring

kaliblazin707, 2019-02-15:
Here is a dump of the matek f405 board with same issue
the fact i have 3 boards with the same issue is driving me insane thinking im doing something wrong..

# dump

# version
# INAV/MATEKF405 2.1.0 Feb 10 2019 / 16:38:02 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 923 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

kaliblazin707, 2019-02-15:
heres a pic showing reciever working in betaflight
https://imgur.com/a/xjFv3bK

digitalentity, 2019-02-15:
Unfortunately I don't have any IBUS TX/RX here to verify.

shellixyz, 2019-02-15:
@digitalentity I already tested IBus on a Matek F405-STD with a IA6b receiver yesterday. It is working. So this is strange. The X6B should work the same. I will test in the evening.

kaliblazin707, 2019-02-15:
No need. please close this thread when you receive my THANK YOU!!!! lol. Turns out.... Im an idiot... The 5v I had the receiver connected to, in all 3 instances, required lipo to activate. I killed the LED on all my x6b's(Its too close to the nylon nut so i scrape it off all of em) and couldnt tell that it was not getting battery voltage. It worked in betaflight because it had lipo voltage when i tested it. Thank you so much for you patience bro. Cheers.

shellixyz, 2019-02-15:
Hehehehe :wink: good

closed 2019-02-15 by shellixyz

digitalentity, 2019-02-16:
Thanks for letting us know!