iNavFlight/inav #4376: Matek F-405 & BN-880: MAG issue

maxledfreak, 2019-02-16:
Hello,
this issue occurs with Matek F-45 and BN-880 GPS/MAG. GPS seems to be work correct, but MAG is doing strange things. When I turn the copter to left or right side, in preview is shown a rotation on Z. Tried different orientation settings ( flipped, not flipped, 180° etc) and many calibration runs but non works correct. In sensor tab, MAG seems to be working correct. 
I've tried to make a video from this issue. 
inav Version is: 2.0.1

Hope anybody could fix this issue.

Greetings from Germany
Max

https://youtu.be/zQvDs-q6NPc

elepete, 2019-02-16:
Max
very unlikely this is a hardware or SW issue. 
I am using a BN-880 as well (on Matek 7222 - but that should not matter)
Looks like wrong orientation: If Antenna patch is facing up and the connector is pointing to the rear end orientation needs to be set to CW270-flip. 
Peter

teckel12, 2019-02-16:
@maxledfreak Really sounds like a classic mag alignment issue to me. I believe the BN-880 is 270 Flip (unless you mount it "backwards" in which case it would be 90 Flip).

leeph, 2019-02-19:
It's also worth noting that the mag alignment has to be set WITH REFERENCE TO THE FLIGHT CONTROLLER.

For the BN-880, CW270Flip is correct when the arrow on the FC is facing forwards. If you've mounted the FC with a rotation, or even upside down, then your mag alignment will be in relation to the FC, not the forward direction of the craft.

For example, if you have done a simple flip upside down of the FC, then the mag should be CW270 (no flip).

If you've rotated the flight controller 90 degrees right, then the mag alignment becomes CW0Flip.

kelleysislander, 2019-04-02:
So Glad I ran across your comments here.  I figured the CW270Flip the hard way!  I really was expecting the alignment arrow to be factually correct, so I approached it from that angle first.  It wasn't until nothing worked then that I began to amass enough observations to try different things.

For me the most important thing here is from leeph:  

"...the mag alignment has to be set WITH REFERENCE TO THE FLIGHT CONTROLLER."

Thanks, Man!@@

Cheers,  Bill

closed 2019-04-09 by DzikuVx

maxledfreak, 2019-04-28:
Hello,
I have tried different orientation and improve calibration. Now it works! 

Many thanks for support! 

Max

SeanTheITGuy, 2019-09-07:
Thank you so much for this thread. I was having a similar issue where the preview looked ok, and POS HOLD worked, but trying to do RTH or WP Navigation had the quad diving off in some random direction at mach 2.  I have my 880 mounted with GPS patch up, and the connector coming out the right hand side (looking from the rear of the craft) so CW180FLIP.  Will test again when the hurricane passes and hopefully no more panic moments trying to regain control.

datamann, 2019-09-07:
@SeanTheITGuy In INAV, in the setup tab where you can see the model moving when you move the craft, does the craft yaw when you pich up or down? If it does the alignment is wrong. Just play with alignment until the craft is perfectly still, no yawing, when pithing the craft. Thats how I fixed my RTH problem.