iNavFlight/inav #4377: How to get Feedback for Stick Command Battery Profile Selection

b14ckyy, 2019-02-16:
Hi Team,

I'm using 2 different batteries on my Wing right now and soon I will also try Li-Ion Packs.

Right now I have 4S 2200mAh and 4S2P 3000mAh. For Li-Ion I want to use 4S2P with 5000 or 6000mAh. So to get proper battery level monitoring, I need to switch battery profiles on the field. 

The INAV Wiki says I can select the battery profile by stick command but I get no feedback if it really selected it. I activated Profile Selection with stick command (But I'd like not to because I only have one profile) and this gives me a beep confirmation. Does Stick command Battery Profile selection even work? To go through the OSD is a bit of a hazzle.

teckel12, 2019-02-16:
@b14ckyy The last time I used it, it worked fine. Not sure if it beeped or not, but it would be a good idea to beep for confirmation if it's not already doing that.

I'll test for the beep, and confirm it's setting the battery profile.

shellixyz, 2019-02-16:
In theory it should beep the same way as with PID profiles. I think it is 1 beep for profile 1, 2 for profile 2, etc. The code haven't been modified but I'd have to test it to be sure it is still working as intended.

b14ckyy, 2019-02-17:
I can only confirm beeps on profile selection. 1-3 beeps depending on profile. But no beeps and no OSD feedback on Battery profile selection. Also the percentage of the used battery mAh did not change (should change if it has lost 10% in disarmed state and change between a 2200 and 3000mAh config. So I assume that it has not changed the profile at all.

teckel12, 2019-02-17:
@b14ckyy The battery percentage could be based on the voltage if it's already used 10%. Looking in CLI is the way to know for sure.

b14ckyy, 2019-02-17:
I know what you mean. I turned it on withj 98% based on voltage. afaik when current meter is on and battery capacity is known, it shows percentage based on used mAh? At least when flying with the bigger battery, it shows 0% battery with 3.6V and with the smaller batteries the percentage reading is correct.

teckel12, 2019-02-18:
@b14ckyy As long as the voltage is within some kind of full range, it will show battery percentage based on mAh capacity.  But, if the battery voltage is is not within the full range, it will show the battery percentage based on voltage.

This kinda makes sense.  If you turned it on with a battery at only 3.8 volts you wouldn't want to show that it's 100% full and assume you have the full mAh to fly with.

Also not sure what happens when you change a battery profile after the battery is partly used.  It may trigger the "not full" logic and use the voltage instead of mAh.  I could see issues with changing the battery profiles mid-battery.

I'll just test this, it isn't too difficult.

shellixyz, 2019-02-18:
> Also not sure what happens when you change a battery profile after the battery is partly used. It may trigger the "not full" logic and use the voltage instead of mAh

Yes that's what happens. When you select a profile it checks if the battery is full or not based on the voltage and if it is not the estimated percentage is linear between `min_cell_voltage` and `max_cell_voltage`

About the threshold voltage a quote from the battery documentation:
> For the battery to be considered full the mean cell voltage of the battery needs to be above vbat_max_cell_voltage - 140mV (by default 4.1V). So a 3S battery will be considered full above 12.3V and a 4S battery above 16.24V. If the battery plugged in is not considered full the remaining battery capacity OSD item will show NF (Not Full).

teckel12, 2019-02-18:
@shellixyz That's what I assumed.  So changing battery profiles could appear to not change anything (as the percent full could be based on voltage not mAh).

b14ckyy, 2019-02-18:
OK thanks for that clarification. But there is still the issue that I don't get any feedback if the battery profile is changed or not. 
Was anyone able to test this?

shellixyz, 2019-02-18:
I just checked it is working fine for me as I described it in my earlier comment. Beep count matches the profile ID. iNav 2.1.0-RC3

shellixyz, 2019-02-18:
There are two ways you can check if the battery you plugged in is considered full or not:

1. Add the remaining capacity element to your OSD. If the battery wasn't considered full when you plugged it in it will show NF
2. In the configurator's setup tab in the `Info` frame

[image]

b14ckyy, 2019-02-19:
> I just checked it is working fine for me as I described it in my earlier comment. Beep count matches the profile ID. iNav 2.1.0-RC3

Ah I thought you are speaking of the Config Profile and not the Battery profile. 
I'll check again tomorrow. I'ts 11:30 PM here in Germany and I need some sleep ;)

Does the TX Stick Command Profile Selection Feature need to be turned on to also select the battery profile?

shellixyz, 2019-02-19:
Yes you do need that feature do select both the PID profile and battery profile with stick commands

b14ckyy, 2019-02-20:
Hey Guys. I tested it again now and still not working :( the PID profile works fine with the beeps but Battery profile shows no reaction/feedback. 

If I switch the batt profile in configurator, the buzzer makes a single beep on every profile but switches battery capacity correctly in config tab. 

see here: https://www.youtube.com/watch?v=FHticmAlHPE

And my Dump:
[MATEKF405SE 2.1.0 RC3 - AR Wing - Final.txt](https://github.com/iNavFlight/inav/files/2886124/MATEKF405SE.2.1.0.RC3.-.AR.Wing.-.Final.txt)

shellixyz, 2019-02-20:
Does the battery profile change with stick commands ? If you are connected to the board with the configurator you can see it change immediately. I have the hunch that your throttle just doesn't go high enough to be considered high, the profile doesn't change so it doesn't beep. Make sure your throttle channel goes above the `max_check` setting when you push the throttle to the max.

b14ckyy, 2019-02-21:
Just tested that again. Throttle value is fine:
[image]

max_check is at 1950 I think. 

But I now tested in configurator. Indeed it changes the battery profile fine (good to know). Buzzer on USB is disabled so I just tried immediately after removing the cable. As I found out now, I need to select the battery profile twice, then I get the beep!!

So selecting profile 2 with gesture > no boop, selecting 1 > no beep > selecting 1 again > 1 beep selecting 2 > no beep > select 2 again > 2 beeps. Then it works fine.

teckel12, 2019-02-21:
@b14ckyy I just tried it and while the configurator showed the profile and battery changing, I got no beeps at all. I'll make sure I'm running 2.1.0 RC3 and try again.

teckel12, 2019-02-21:
NEVERMIND! After flashing 2.1.0RC3 I then realized the problem, I had `MULTI_BEEPS` turned off. So of course it wasn't beeping. I retested and it beeped correctly each and every time, if it was the first time or not, for both profile and battery. You do need to hold the sticks in the correct orientation for a second or two. But it all worked exactly as expected, beeps and profiles changing according to the configurator.

b14ckyy, 2019-02-22:
Okay that's strange. Did you check my video? It was reproducible that I needed to do the stick command 2 times to the same profile until I got the beep. On first time there was no beep at all and I have done it very carefully but it was reproducible.

teckel12, 2019-02-24:
@b14ckyy I did see the video, and mine does not operate like that. My F722SE beeps the correct amount for all profiles every time.

teckel12, 2019-02-24:
@b14ckyy Do you happen to have the feature PWM_OUTPUT_ENABLE turned on? There's 3 issues having to do with beeping on the F722SE and I'm trying to find a similarity as my F722SE beeps correctly. One issue has DASHBOARD turned on and another has RSSI_ADC turned on. Looking for differences in my setup.

closed 2019-04-27 by shellixyz