iNavFlight/inav #4379: Cruise Direction not Reliable? 2.1-RC3

b14ckyy, 2019-02-17:
## Current Behavior
Sporadically GPS Heading is freaking out and causing cruise direction change. 
Not sure if this is a config or GPS module issue or a bug but I want to report it. 

## Steps to Reproduce
1. Fly in cruise mode
2. Often change direction
3. Watch if direction is kept when loosing sticks
4. Sporadically Cruise changes direction from alone extremely

## Expected behavior
Cruise Mode should keep direction of the moment, when Pilot releases sticks.

## Suggested solution(s)
Not sure, if it is a bug or a misconfiguration. 

## Additional context
Links to a Screen record of Blackbox replay, Dump and Log.
[LOG00036.zip](https://github.com/iNavFlight/inav/files/2872477/LOG00036.zip)
[INAV_cli_20190217_001111.txt](https://github.com/iNavFlight/inav/files/2872480/INAV_cli_20190217_001111.txt)

https://www.youtube.com/watch?v=j3CfgngOVzo

---
- FC Board name and vendor: Matek F405SE with UBLOX M8N GPS and HMC5883 Magnetometer
- INAV version string:
# INAV/MATEKF405SE 2.1.0 Feb 10 2019 / 16:44:59 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

giacomo892, 2019-02-17:
@stronnag I've checked attitude[2] and gps course over ground but cannot find why MWP displays the wing flying sideways. Can you enlight me about the bearing/heading and machine icon display in MWP?

stronnag, 2019-02-17:
@giacomo892 let me see ...

giacomo892, 2019-02-17:
@stronnag you can use the log attached to the issue, but i'm not saying it is an issue with MWP, actually inav was displaying a wrong heading icon on OSD and GPS assisted modes aren't working 100% fine.

stronnag, 2019-02-17:
Ah, I thought you meant mwp shows the icon incorrectly. 
I fast-forwarded the whole log. Seem to be fine *apart* from a couple of minutes (or so) period around 37 - 38 minutes when this happens. After the subsequent F/S it sorts itself out again.

[image]

At this point, th e machine continues in N direction (showing W'ly icon heading) for another 30s before the F/S.

Seems to me consistent with the reported problem.

stronnag, 2019-02-17:
c. 8x real time video : https://youtu.be/tVI71HgafOU

See the flight timer (lower right) around 37-38 minutes, where the plane does the impossible.

Note also the Heading v. GPS CoG (Upper right). Here
* Heading is the attitude[2] value from blackbox
* GPS CoG is calculated by mwp from consecutive GPS positions.

The CoG agrees with the track, while the icon heading is attitude[2].

b14ckyy, 2019-02-17:
> @stronnag you can use the log attached to the issue, but i'm not saying it is an issue with MWP, actually inav was displaying a wrong heading icon on OSD and GPS assisted modes aren't working 100% fine.

Exactly. During Cruise Flight the plane suddenly changed direction but without stick input but I'm able to counter control that manually, staying in cruise. Never saw this in RTH or FS-RTH. Only in cruise this seems to happen but I have not tested Mission flight yet. As stronnag explains, It seems to be a issue with the GPS self-calculated heading? (Compass is not used here AFAIK) So could this possibly be an issue with my GPS module? I only recognized this that hard on this particular flight. Today I ran another battery and no direction issue on 25min flight. I'll upload the Log anyway for reference. 
[LOG00039.zip](https://github.com/iNavFlight/inav/files/2872962/LOG00039.zip)

I will drive back in the city today so It will be difficult to do further test flights.

digitalentity, 2019-02-17:
I have received a few reports of misbehaving RTH and Cruise mode at location - Eastern Russia and time - yesterday. It was firmware 2.0.1 though. My theory is that we had some disruptions in GPS, GLONASS or BEIDOU that caused temporary disruption of calculated data.

b14ckyy, 2019-02-17:
I have a very cheap M8N module that AFAIK only uses gps and not GLONASS or BEIDU. that's why I always only get 12 sats at peak but get them very fast. Could be a good explanation of this. Will watch if it happens again or if other people report this.

OlivierC-FR, 2019-02-17:
I have similar issues with GPS / wrong heading, I post here because it may be related :

digitalentity, 2019-02-17:
@b14ckyy all M8N modules are capable of at least GPS+GLONASS. Little satellite count on M8N is usually caused by crappy antenna used there. We've already seen glitches with GLONASS a few times.

Interestingly, we didn't change anything in heading calculation internally.

stronnag, 2019-02-17:
I made another replay video, this time instructing mwp / bbl replay to use the blackbox GPS CoG as Heading (while mwp's GPS COG is again calculated by mwp).

This time the plane's heading is consistent with motion.
https://youtu.be/lsY6e9voJRM

So a mystery, probably not explained by the military practising GPS denial on unsuspecting model aviators.

b14ckyy, 2019-02-17:
Big mistery indeed. But after the vids the other user made, I'm a bit concerned about RTH on a failsafe situation where i possibly cannot take over. If the wrong heading takes longer than 30s, it is likely to trigger the emergency landing threshold when moving away 500m during RTH. Is there a way of a sanity check? Like... if CoG heading is too far away from position change direction, then switch to software calculated direction instead, though it is slower? Or would you recommend to go completely to the software calculated direction?

b14ckyy, 2019-02-24:
Hey guys. I was flying again and today it was close to the worst thing that could happen. I got a failsafe because of a bit range flight and at this point the CoG was completely off by 180°. Instead of going home, the plane started to circle, to get the correct direction that was never reached. Fortunately I gained back control by raising the controller and was flying back in cruise manually but had to correct the course the whole time. MWP showed the plane flying backwards the whole time. I begin to loose trust in the auto navigation :-/ a few hours before I tested the range of my helical FPV antenna and went out 4.7km and then got a failsafe but there all went fine. 

Could this be a faulty GPS module? But I'm not the only one recognizing this kind of issue :-/

LOG: 
[LOG00061.zip](https://github.com/iNavFlight/inav/files/2898248/LOG00061.zip)

Points where it can be seen:
- during Launch
- 4:25
- 5:10
- 10:10 < the big failsafe failure

stronnag, 2019-02-24:
I doubt it's a bad GPS; if you tell mwp to use GPS CoG for 'heading' then the model is visualised correctly with respect to the track.

When you tell mwp to use attitude[2] (by default), then the visualised heading behaves strangely as you've noted. Looks to me like iNav is somehow getting the heading very confused.

giacomo892, 2019-02-25:
@stronnag  That's why #4390 has been introduced.
Fixed wing pilots that uses mag will figure out very soon if their setup is not configured correctly after this PR

b14ckyy, 2019-02-25:
> @stronnag  That's why #4390 has been introduced.
> Fixed wing pilots that uses mag will figure out very soon if their setup is not configured correctly after this PR

Not sure if I understand that correctly. Till now afaik, the GPS is not used in flight. Only during launch to launch the direction it was armed. That's why I have activated it to make sure to launch in a straight direction and that worked fine for me. After that change, the plane will use mag heading instead of GPS CoG, when available?

> I doubt it's a bad GPS; if you tell mwp to use GPS CoG for 'heading' then the model is visualised correctly with respect to the track.
> When you tell mwp to use attitude[2] (by default), then the visualised heading behaves strangely as you've noted. Looks to me like iNav is somehow getting the heading very confused; however, I note you have the compass enabled --- what happens without that?
> H mag_hardware:2

I could disable it the next time. But I can only fly again on the weekend.

giacomo892, 2019-02-25:
@b14ckyy it's the exact opposite with the version you are running. MAG is used on the ground and GPS heading while moving. If the MAG and GPS heading diverges too much during flight then INAV used MAG heading which in your case could have been wrong. Wait the final 2.1 and test again in few days.

b14ckyy, 2019-02-25:
Oh wait! I just remembered a change but I'm not sure when I have done this. Need to change my older logs. 

Could it be possible that a too high baud rate causes GPS CoG to freak out? Anytime during the last weeks, I think when I switched from 2.0.1 to 2.1 RC2, I changed the GPS baud rate from default 38k to 115k baud rate. (Isn't 115 default since 2.1 either?).
 So I will check all my logs today when I'm home, to see when I have changed that and if it was happening before or directly after that.

giacomo892, 2019-02-25:
115200 is perfectly fine and it is suggested for the 10Hz UBLOX7 mode.

b14ckyy, 2019-02-25:
OK good. Then I'll wait for the Final release with the potential fix. Itherwise, if you can send me a dev version for the F405 wing (I have no clue how to compile for myself) I could test it asap.

giacomo892, 2019-02-25:
Sure. 
[inav_2.1.0_MATEKF405SE.hex.zip](https://github.com/iNavFlight/inav/files/2899755/inav_2.1.0_MATEKF405SE.hex.zip)

b14ckyy, 2019-02-26:
Thanks for the firmware. I have updated yesterday and in the evening the final 2.1 was relased. Is this change in the final too? Then I would use this and transfer all my settings from the RC3. If not, I will stay with your version for testing.

giacomo892, 2019-02-26:
Final has the "fix". Be sure to have your compass calibrated and well oriented and everything should work.

b14ckyy, 2019-02-26:
Great thank you. Compass orientation is precise to about 1° or 2°. 
btw a short side question. Is it possible to set a alternative compass orientation in the CLI to, let's say 45°? 

In GUI I can only select 90° steps and flip. My GPS has a Compass with 45° turn and in needed to mount it in the correct 45° angle to have a correct direction and it works fine. But... I plan to buy a Original M8N GPS soon and there the orientation is straight. But I want to avoid to cut another whole in the wing for it. So it would be great to mount it in the existing 45° angled opening and just set it to 45° in CLI. :D 

[image]

giacomo892, 2019-02-26:
Yes you can in the CLI, you can input whatever you like.
So be sure compass points the correct direction also when you apply some throttle and you tilt the plane and you should be good to go with 2.1 and mag enabled! Let us know. Don't take risk if the home arrow is not pointing the correct direction after the launch

b14ckyy, 2019-02-26:
Great. Will test on Saturday. Hope the weather will be fine. And Power Cable Mag-Distortion shouldn't be a problem. 12A is max current and it has 20cm distance from any power cable :) Thanks for your support!

b14ckyy, 2019-03-13:
Still no test flight weather. Storms up to 90km/s and my wing won't be happy :(

mike-teel-NEFL-FPV, 2019-03-27:
I'm having similar issue f35 f4 FC 1.9 RTH looses it's mind...here's easiest way to explain...
https://youtu.be/a2XJ6HSucaU

mike-teel-NEFL-FPV, 2019-03-28:
Giacomo892 updated

b14ckyy, 2019-05-05:
Hey guys,

just wanted to give you a last update. I have flown many hours now since the release of inav and using the compass for navigation. Never had a flyaway again. The only downside is the bad precision of the compass in my GPS module. So it returns in a S-Shape sometimes and also cruise is not as straight as I'd like but at least the RTH and Failsafe is reliable enough to come back. So the change seems to work well. 

Maybe I will replace the GPS anyway. In the future, I will only use genuine M8N modules and no cheap china crap again.

b14ckyy, 2019-07-07:
Fixed as we can decide for GPS or compass heading now.

closed 2019-07-07 by b14ckyy