iNavFlight/inav #4382: Quad took off after RTH landing.

krzysztofmatula, 2019-02-17:
What happened:
- RTH completed and the quad safely landed (OSD was displaying LANDING and then LANDED)
- just after that the motors slowly started to increase power
- eventually the machine took off!

See very last part of this video: https://www.dropbox.com/s/t3l10ct934hfjzn/PICT0018.AVI?dl=0
And the corresponding log: https://www.dropbox.com/s/r3dg3kyz6czdq6m/LOG00012.TXT?dl=0

I think it is nothing new. I remember I was observing increasing motor power previously but it's first time when it actually made the machine take off!

Looking at the log is seems the RTH is commanding min throttle (1050). Some stabilization code must still operate and increase the motor rates. As a safely manner I think it should be blocked after entering the LANDED stage, which was detected correctly.

b14ckyy, 2019-02-17:
Your compass is badly aligned or not calibrated exactly. That's why it is toilet bowling in the first place. so when touched down, the copter still tried to correct it's position (position correction always active when in auto mode) so it touched the ground with an angle and trying to raise the angle even further to correct for the position mismatch. The I term rises up further until the thrust lifts the copter again. 

So calibrate you compass and I recommend to enable nav_disarm_on_landing, to turn it off on touchdown.

krzysztofmatula, 2019-02-17:
Toilet bowling is one thing, but the final descent was smooth.
GPS+MAG are just on the LiPo. Not perfect placement, but with no high throttle MAG works OK.

The issue is that after detecting landing it still tries to do adjustments as if it were flying. I suggest inaw don't do this regardless of having nav_disarm_on_landing ON or OFF.

b14ckyy, 2019-02-17:
No the compass IS the problem here. 

As you say the compass is affected by the current flow. on hover throttle it might be "acceptable" (but how it is circling in the video I wouldn't even consider to continue auto landing). 

Looked into the log and here is a video. Short before touchdown it looks okay, besides of the circling that's not okay. 

https://1drv.ms/v/s!Ar6QsGgoXRgqm-t3EVK_K50FSeLybg

then touchdown > throttle goes close to zero > curent flow drops > magnetometer influence changes and the copter "thinks" it has not the correct heading and tries to correct that. That's why top left and bottom right motor do spin up first to turn left ways and it rises up further and further because there is no reaction on the ground. Also it tries to get it's final position. 

If you auto-land without automatic disarm, you possibly want to launch again so it makes no sense to disable the PID controller. You MUST disarm the copter or you will have the same issue on every landing if you refuse auto disarm. The main issue still is your compass here. Fix the position and the issue will be gone. Had the same issue on another copter a year back and after placing t correctly, all is fine.

b14ckyy, 2019-02-17:
ah and btw... See how the compass is totally freaking out depending on the sum of motor throttle? the copter was not even turning during descent so all lines should be flat!

[image]

[image]

[image]

krzysztofmatula, 2019-02-17:
Oh, come on, those are really small anomalies :)
The machine successfully returned and landed after all.
See how little they influence the calculated heading. Maybe 3 degrees, not more.

Also see that during the questionable period - after LANDED - MAG readings are stable, while motors 1 and 4 spin initially at 40% and accelerate. I think the issue could be that it happened the machine landed with slightly different heading that the one expected. Could be even caused by uneven grassy surface. Then the controller tries to rotate it *while it has landed*, eventually causing take off.

teckel12, 2019-02-17:
@krzysztofmatula You could just set it to turn off the motors after landing. Then, if it lands on uneven ground it won't try to stabilize itself and take off in the process.

b14ckyy, 2019-02-17:
> @krzysztofmatula You could just set it to turn off the motors after landing. Then, if it lands on uneven ground it won't try to stabilize itself and take off in the process.

That's what I told him before but he won't do that ^^ 

I checked in MWP tools. At touchdown the heading instantly changed by 10°. That's enough that the controller tries to compensate and tries to correct his heading. Also if landing is complete and you do not disarm, it will automatically switch to last used mode, So Pilot Control is reactivated and your last mode was AIRMODE. 

So I'll repeat my suggestion: Fix your compass or enable disarm on land. ;)

b14ckyy, 2019-02-17:
> Update: while arming the heading was 252deg. After landing it happened to be 232deg. When it has landed 20degrees off the course it shall stay as is and not try to turn on the ground. That's what this issue is about.

Nope. You are referring to the bearing, The last GPS moving direction. That doesn't matter because the copter needs to know it's heading to keep its possition, not it's moving direction. The compass heading is relevant here. 

Compass heading was 52° on takeoff and 52° on landing phase (what it has measured). on touchdown it jumped then from 50° to 60° (clockwise) and that's exactly why motor 1 and 4 spin up to turn the copter counter clockwise.

stronnag, 2019-02-17:
> So I'll repeat my suggestion: Fix your compass or enable disarm on land. ;)

Or preferably both; in the replay there is a consistent difference between heading and CoG.

krzysztofmatula, 2019-02-17:
@teckel12 sure I could :)
In fact I don't actually need the landing at all. I was just testing it to help make the software better.

@b14ckyy thanks for suggestions but I really don't need it to work well for my setup.

Actually I start disliking the tendency of persuading that all problems are user-errors and that every one shall build perfect machines, then start touching inav... :(

b14ckyy, 2019-02-17:
> @b14ckyy thanks for suggestions but I really don't need it to work well for my setup.
> Actually I start disliking the tendency of persuading that all problems are user-errors and that every one shall build perfect machines, then start touching inav... :(

Please don't understand me wrong. I totally get your point and some kind of extra safety feature is always great. In a perfect world we could have a software where we just plug everything on how we want and the software will figure it all out by itself how it has to act. 

But in reality that's not how it works. In the Wiki of inav there are clear instructions how to mount the compass correctly, far away from high current wires and interferences. If the wires influence the magnetic field, it is impossible to get a proper heading out of the sensor. No matter how good the software is. 

And the feature you asked for, to disable the automatic correction, makes not much sense because to disable the PID controller you have to disarm the copter. That's how it works. Why would someone keep the motors spinning after landing if he can't takeoff anyway because of the disabled controller? 

Arducopter has a feature to detect magnetic interferences on different throttle values. It helps to compensate for that a bit but only for slight influence. It will refuse to accept it if the interference is too big. That could be a possible feature request for inav but testing a big copter with 10" blades on the bench with full throttle is a very dangerous thing anyway. So I'd not recommend to do that…

krzysztofmatula, 2019-02-17:
Summarising the observations, the current state is that:
- if it happens the MR lands 10degrees off-course (from whatever reason) it may happen it will automatically takeoff after a while.

If we agree this is fine - the issue may be closed.

If not, there are several options:
1. clearly mark in doc that flying without auto_disarm is dangerous
2. force the auto_disarm always on and remove the option
3. make sure that after entering LANDED state motors are only idling

I opt for 3.

teckel12, 2019-02-18:
@kylemanna 
> disliking the tendency of persuading that all problems are user-errors and that every one shall build perfect machines, then start touching inav... :(

I was just suggesting using a setting that maybe you were not aware of which disarms once landed.

kardon18, 2019-02-18:
Check out time index 2.30 on the video. The home arrow drifts by 180degs as the quad slows down and turns.  
Then once it starts moving forward again, it then appears to use the GPS as a directional reference, and re-aligns the home arrow.

teckel12, 2019-02-18:
@kardon18 This is a compass alignment, calibration and/or interference issue at its core. Can't stress enough that you shouldn't be relying on RTH or position hold until the compass is sorted out.

krzysztofmatula, 2019-02-18:
> This is a compass alignment, calibration and/or interference issue at its core.

I don' agree.
This is what helped to discover the issue that after landing it may happen to takeoff again :)

I think here's the fix (to be verified): https://github.com/iNavFlight/inav/pull/4383

teckel12, 2019-02-18:
@krzysztofmatula I agree that there's two issues here.  One, there's a compass calibration/alignment issue.  And two, once landed it shouldn't take off again.

krzysztofmatula, 2019-02-27:
Fixed on 2.1.

PR #4383 (2019-02-25): Prevent unexpected takeoff in RTH LANDED state.
This should fix #4382.
Not tested yet.
closed 2019-02-27 by krzysztofmatula