iNavFlight/inav #4387: Servo Mixer odd layout

dankvwguy, 2019-02-18:
## Current Behavior
When using versions of iNav above 2.0.0, the servo mixer page gets all out of order with random values and inputs. it doesnt appear to affect anything but in order to make any changes, i have to load up older version of iNav. anyone else start experiencing this issue? Ive included a picture of the 2.1.3 mixer page
[image]

## Steps to Reproduce

1. Run any version of iNav above 2.0.0
2. Open Mixer page
3. Look at Servo Mixer box
4.

## Expected behavior
The servo mixer page should appear as it does in older version of iNav

## Suggested solution(s)
No idea how to solve the issue

## Additional context

---

# version
# INAV/MATEKF405SE 2.1.0 Jan  8 2019 / 13:34:44 (f8de2e6a3)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

- FC Board name and vendor: Matek F405SE
- INAV version string: 2.1.0

Arvycka, 2019-02-18:
Use Inav 2.1 RC3 with the latest configurator. 
Use configurator 2.0 for Inav versions previous to 2.1 RC3

shellixyz, 2019-02-18:
Working fine for me, I can't reproduce the issue. Tested latest configurator (to be released soon, 8792188c) with version 2.0.0, 2.0.1 and 2.1.0-RC3

shellixyz, 2019-02-18:
Note that you can't use configurator version 2.1.3 with FW versions 2.1.0-RC1 and 2.1.0-RC2

dankvwguy, 2019-02-18:
ok, good to know guys. that must be my issue. ill change FW or just use the other configurator for now. just wanted to make sure it wasnt a major issue. THANKS!!!!

closed 2019-02-18 by dankvwguy