iNavFlight/inav #4389: Naze32 won't connect to configurator 2.1.2

bobkat1, 2019-02-18:
I have 3 Illuminati32 flight controllers loaded with Inav 1.7.3 and for some reason none of them will connect to the configurator. I have not tried to connect them for a while and now they don't work. I'm still using the same configurator I have used all along and it always worked great. Any idea on why it won't work now? I know there's a newer version of the configurator and newer firmware for the controller, but as far as I know the Inav 1.7.3 is the last version that will work on this board. I did try reflashing 1 board with an old version of cleanflight and that was able to connect up fine with the cleanflight 1.2.4 configurator, but when I flash the Inav 1.7.3 on it agian it won't connect.

stronnag, 2019-02-18:
Naze32 is not supported
You need a contemporaneous configurator (1.9 at the latest).

closed 2019-02-18 by stronnag

bobkat1, 2019-02-18:
The Inav 2.1.2 is the same configurator I used to flash Inav 1.7.3 to these boards in the 1st place and was able to configure them with no problems 6 months ago so that makes no sense that it won't work now.

stronnag, 2019-02-18:
Given that iNav Configurator 2.1.2 was only released Fri Feb 8 22:16:57 2019 +0100, you're somewhat misinformed.

* This is not a support channel
* Naze32 is not supported, so this is my last word on the matter. Good luck.

bobkat1, 2019-02-18:
I did download the 1.9.3 version and it does work. I have no idea how my configurator could have been updated to the newer 2.1.2 version, but it seems that is the case. I must have been using the 1.9.3 when I set these up and some how my configurator got updated on it's own as I did not update it. I do appreciate the help on this as it looks like everything is working now, but I'm still confused on how my configurator was updated without me knowing about it.