iNavFlight/inav #4391: Better Follow Me Support

janbolt, 2019-02-18:
Ez-gui is dead and mission planner for inav does not implement follow me because of insufficient inav support. (Seems to be a heading problem).
[https://www.rcgroups.com/forums/showthread.php?3030784-Mission-Planner-for-INAV-%28Android%29/page3](url)

Please add sufficient support so we will get follow me some day.

Pairan, 2019-02-19:
maybe you find it in these topics:
#3626
#4112

teckel12, 2019-02-19:
@janbolt With TBS Crossfire full TX module it supports Bluetooth. This can be used to send MSP via Bluetooth from your phone to your transmitter and then to your RX and flight controller (and back). With this already existing two-way communication, the GPS location from your phone could be sent to your model (with of course a simple mobile app to forward your phone's GPS coordinates via MSP to the FC).

Once the FC knows where your phone is (what it's going to follow), it's simple math to turn in the correct direction and maintain a certain distance.

I've never considered a follow me mode in INAV (mainly because I have a DJI Mavic that has the feature which I've never used).  But, off the top of my head, that sounds like the easiest way to implement a follow me mode in INAV. It would require no unique hardware (just a full Crossfire TX module).

janbolt, 2019-02-19:
yes, follow me could work in general. But we don't have an app anymore.
Ez-gui is dead and mission planner for inav didn't implement follow me.
The reason ezio (the developer of mission planner for inav) didn't implement follow me
is the insufficient inav support.
In the thread [https://www.rcgroups.com/forums/showthread.php?3030784-Mission-Planner-for-INAV-%28Android%29/page3](url) ezio is asked for follow me support. His answer is:

"I didn't implement this because currently there is no way to control heading or set point of interest. I think follow me without keeping the model facing you is pointless. I Will be glad to see this added to inav. Maybe if larger amount of people ask about it the inav devs will finally implement it."

So I wish inav would support all needed features and we will get follow me in mission planner for inav.

lgojniczek, 2019-02-19:
Hi, if you are interested in some kind of Multiplayer mode to help everyone to find each other and create some kind of formation I could help as I am an author of Android application named TrackMyRC². It is still in test phase but I'm going to release it soon as it passed all the test successfully.

You can see it in action here https://www.youtube.com/watch?time_continue=39&v=diaqSy9UehY and here https://www.youtube.com/watch?v=fO28dZgoC7Q

It supports eLeReS, Pitlab and is almost ready to support native FrSky telemetry protocol (is ready to support more proto as it has abstract telemetry layer).

Feedback is really appreciate

teckel12, 2019-02-20:
There's some kind of confusion here, and I'm not sure if it's with me or others. But, what I think of when "follow me mode" is mentioned is that the quad follows a person or moving object on the ground. Sometimes this is done via optical object tracking (like with DJI). And other times it's done by simply tracking another GPS device (a phone, transmitter, or tracking device).

With INAV, the optical follow me is out of the question. So follow me mode would typically be done by tracking the pilot's phone GPS position which is transmitted to the model which moves to a waypoint at a particular yaw as to keep the same distance and facing towards the pilot.

Anyway, that's what I'm thinking this issue is about. If it's about one model following another model, that probably shouldn't be called follow **ME** mode as we're not in the model. Me is with the transmitter on the ground.

So who's confused here?

janbolt, 2019-02-20:
@teckel12: you are right. Follow me means an app is sending waypoints and heading via msp link and let the modell follow the device running the app (smartphone).
[http://ez-gui.com/manual/followme/](url)

@lgojniczek: that sounds interesting but that's a different topic.

teckel12, 2019-02-20:
@lgojniczek For sure interesting and very cool, but not "follow me mode".  If you're using this term, you should probably call it something else to avoid confusion.  See DJI for examples of using the term "follow me mode" for a few years.  And I believe Hubsan or another company had a "follow me mode" even earlier.

stronnag, 2019-02-20:
> So who's confused here?

I think you are. The idea is that your machine follows you on the ground, from a phone's GPS location. In order to implement this the way he desired, Bart (ez-gui) was referring the the MSP SET_HEAD and SET_POI directives in # 154.

janbolt, 2019-02-20:
yes, this was covered in #154 almost 3 years ago.
In my understanding the missing SET_HEAD and SET_POI implementation is what Bart stops from implementing follow me in mission planner for inav.
So please implement it so we get follow me (Arducopter can do it for years).

fiam, 2019-02-20:
@janbolt INAV is open source and anyone can open a PR. If you guys want better support for a specific use case or feature there's nothing better than writing the code and submitting it.

Closing as duplicate of #154

closed 2019-02-20 by fiam

teckel12, 2019-02-20:
@stronnag 

> > So who's confused here?
> 
> I think you are. The idea is that your machine follows you on the ground, from a phone's GPS location. In order to implement this the way he desired, Bart (ez-gui) was referring the the MSP SET_HEAD and SET_POI directives in # 154.

Now I think you are confused because that's exactly what I've been referring to as follow me mode.  Others here are talking about a follow mode where one model tracks or follows another model in the air.

georgekucher, 2019-08-24:
How about altitude estimation for FOLLOW_ME mode. In case you are going up or down the hill? Does it use GPS altitude from EZ-GUI or we need a rangefinder connected?