iNavFlight/inav #4392: Matek f722 wing current sensor giving bad values

benjinne, 2019-02-18:
## Current Behavior/Steps to Reproduce
At the default scale (250) and zero offset, the reading is acceptable at idle, then when increasing the current to 1 amp using a clamp meter, inav reports about 1 amp. Increasing till the clamp meter reads 2 amps, inav reports about 15amps, also the values are changing significantly (.5v) when no changes in throttle are present, and the current meter doesn't show changing current. By increasing the scale, the current doesn't increase as rapidly, however the scale had to be adjusted to 6000 which seems really large to me. Also when I increased the current consumption to 5 amps (reading from amp meter), inav reported 1.7 amps. It seemed like some sort of overflow occurred and the reading went back down, or something is really wrong with the current sensor. 

I also made sure the current_meter_type was set to ADC

## Expected behavior
Should be able calibrate the current sensor to give reasonable values

## Additional context

cli dump:
https://pastebin.com/mJqkT8XD

- FC Board name and vendor: F722 Wing Matek
- INAV version string: INAV/MATEKF722SE 2.1.0 Feb 10 2019 / 16:42:36 (0d3d0ff30)

benjinne, 2019-02-21:
Has anyone successfully calibrated their current sensor on the f722 matek wing FC? I want to know if mine is defective or if there's actually something wrong in the firmware.

shellixyz, 2019-02-21:
I should be able to test soon

teckel12, 2019-02-21:
I have the Matek F722SE (same target as the F722 Wing) so I assume the hardware is almost identical.  And I've calibrated my current sensor without issue.  My settings are as follows:

```
set current_meter_scale = 179
set current_meter_offset = 290
```

I calibrate on the bench with props on (quad strapped to cinder block) and use an inline current meter.

ShawnzAZ, 2019-02-23:
From Matek's site:  http://www.mateksys.com/?portfolio=f722-wing#tab-id-2

Battery Voltage Sensor: 1:10 (Scale 1100)
Current Senor: 132A,  3.3V ADC,  Scale 250

teckel12, 2019-02-23:
@ShawnzAZ I never look at manufacturer's current sensor values, they seem to never be right. I calibrate on my own, and they're sometimes FAR off.

MATEKSYS, 2019-02-23:
@teckel12  F722WING has different components and scale of current sense from F722SE
maybe some manufacturers don't know how to count the current scale,  but we always mark right scale according to the components spec in theory.  a bit error is normal due to components value precision.

@benjinne Try old firmware version. http://www.mateksys.com/Downloads/FW/inav_2.1.0_MATEKF722SE_NOV20.hex  from de_pinio branch Nov.20
CLI "defaults" after reflashing

benjinne, 2019-02-23:
Ok so I tried that firmware, but it seems like I have the same issue still. At a scale of 1500, I'm getting resonable readings from 1-5 amps, however the values are very inconsistent, It'll vary +- 1 amp. I'm not sure what's wrong, however I suspect that it could be a noise issue. I'll try adding a capacitor to the esc leads, and report back.

benjinne, 2019-02-23:
So I added a low esr capacitor to the esc lead, and set the scale to 250. This time, I'm getting reasonable values from 1-7 amps! However, I'm still getting +-0.5 amp readings. Seems like either I have a lot of noise, or the ADC for the current sensor is having trouble. In case anyone is curious, I'm running a flycolor 30A BLS-Pro esc with a 2216 sunnysky and a 10 inch prop. I also added a 1000uF 35 Panasonic cap .

benjinne, 2019-02-23:
I updated to latest firmware (F722SE RC3), and I had to double the scale to 500. This time, it seems like I'm getting very accurate values and it doesn't seem to varying as much as before. Maybe +-(0.1-0.2 ) Amps

teckel12, 2019-02-23:
@benjinne Adding the cap is a good idea anyway, so the exercise had multiple good results.

benjinne, 2019-02-25:
Thanks, anyways I realised I made a mistake. The scale actual led turned out to be 260, and not 500. I'm not sure how I got that value, but just wanted to let anyone who might have a similar issue know where I'm at.

teckel12, 2019-02-25:
@MATEKSYS In this case, for the F722SE in INAV by default it sets CURRENT_METER_SCALE = 179. So, anyone with a F722 Wing will have a bad default current scale. Maybe there should be a unique target if the hardware is even slightly different.

digitalentity, 2019-02-25:
Let's keep this open. Seems that we need a different target (or board hardware detection) for WING

MATEKSYS, 2019-02-26:
@digitalentity no need to add variant for F722-WING,  The current scale is available in manual and our webpage.

teckel12, 2019-02-27:
@MATEKSYS So only the current sensor is different between the SE and Wing models? If so, I tend to agree that's not enough to worrent a different target. It could, however cause some confusion.

krzysztofmatula, 2019-02-27:
> for the F722SE in INAV by default it sets CURRENT_METER_SCALE = 179

In my F722SE I had to tune the current meter scale from 179 to 218 to get good readings.
So, the defaults are to be treated as initial values anyway.

digitalentity, 2019-03-01:
@MATEKSYS is there any hardware difference between F722 and F722-WING apart from current sensor resistor?

MATEKSYS, 2019-03-02:
@digitalentity  F722WING has no ICM20602 builtin, It works fine with default MPU6000.
more variants will confuse users

Avegawanderer, 2019-03-19:
I have similar issue with Matek F722-wing and iNav 2.1.0.  Increasing throttle gives increase of current up to 10amp, further increasing shows current of about 0-5amp, while after-flight osd summary shows 68amp max, which cannot be reached on my setup. Scale is 250 according to Matek, offset is 0. Seems to be some kind of overflow in software. I'll try to do some tests with static load and check it out if the issue is hardware-related or not.

wx4cb, 2019-03-19:
im not sure about that. @Avegawanderer i was pulling 80 some-odd amps on my new one the other day.

Brand new F722 wing. and those current values are about correct (changed it to the one as per the matek page for the FC) as i know this setup tops out at about 75 amps.

[image]
[image]

Avegawanderer, 2019-03-25:
So I did some tests and I am convinced that this issue is not an iNav bug, but related to Matek board.
Under static load from 0 to 20amps I get stable and correct current readings. I believe, the actual reason for incorrect current readings is the high side current sensor, which possibly is not as tolerant to noise produced by ESC as more common low side sensor, It seems that the problem only arises when there is no low-esr capacitor connected to ESC (or power output from the FC). So I think that if you use an ESC without electrolytic capacitor (like ESCs for racing drones), a low-esr capacitor on power output of the FC is not only a good idea but  a must.
[image]
Current sensor output (thin red wire)

[image]
Current sensor output, throttle 50%, no capacitor

[image]
Current sensor output, throttle 75%, no capacitor

[image]
Current sensor output, throttle 100%, no capacitor

[image]
Current sensor output, throttle 50%, 470uF 35V capacitor

[image]
Current sensor output, throttle 50%, 470uF 35V capacitor

[image]
Current sensor output, throttle 50%, 470uF 35V capacitor

wx4cb, 2019-03-25:
high side current sensor???? there's only 1 current sensor on the board afaik... but that aside, yes it does seem like you have a VERY noisy esc. that's not the fault of the matek board though that's potentially a bad ESC. in that case, yes, adding a cap would help, but it doesn't fix the underlying problem

Avegawanderer, 2019-03-25:
High side current sensor is a kind of current sensor where shunt is injected in supply rail. Low side, as opposite, means that current sensor is injected into GND rail. High side sensors use differential amplifier or current mirror adding some gain to voltage drop on the shunt itself, see for example [https://www.analog.com/en/analog-dialogue/articles/high-side-current-sensing.html](url) 
or 
[https://www.st.com/resource/en/datasheet/tsc101.pdf](url)
I don't know the reason why Matek put high side sensor for this FC, possibly to keep common ground with other power contributors (maybe it's usefull for airplanes).
And yes, I tried other ESC (Turnigy plush 30A), but the noise was about the same (similar to last three pics).
It would be nice to see one's pics from the same point with different ESC and compare them to what I've got.

shellixyz, 2019-03-25:
AFAIK All the FCs are using high side sensing

lun606, 2019-04-14:
I fine the problem...
I change back the stock esc(use one shot) everything will be good.
But why i use Hobbywing X-Rotor 40a dshot esc (even i setup esc to one shot or dshot) the current sensor was not work.

closed 2019-05-24 by DzikuVx

madl3x, 2022-05-15:
Thank you @Avegawanderer! 👍  Based on your findings I was able to fix the current sensor on Matek F722 Wing. I placed a 1000uF 35V capacitor in parallel with the ESC.

My problem was happening on a ZoHD Nano Talon, only when using LiIon (4S @ 3000 mAh Sony VTC6) batteries. The current sensor was all over the place (showing either 0.5A or 50A mid-flight, instead of 3-5A as usual), but below 1A (motor idle) everything looked ok.

What is interesting is that I didn't have the same issue with the LiPo batteries that I usually use for my racing drones (4S @ 1500 mAh Tattu R Line 120C). Probably because these batteries are high-discharge, the spikes aren't happening.