iNavFlight/inav #4394: Firmware for inav 2.1.3

jaimegil118, 2019-02-19:
Downloaded inav 2.1.3 for Mac. Listo of firmara re avala le is 2.o. Series , from Oct 2018. Can't flash controler ( ómnibus f4, matek f405).
I never experienced Thea problems in previous inav versions
Where is updated firmware?
Regards
Jaimegil118

teckel12, 2019-02-19:
@jaimegil118 The latest release of INAV is 2.0.1.  There's a release candidate for 2.1.0 which doesn't download by default with the configurator. If you want to download pre-release versions, you need to specify that in the configurator.

The latest stable release can always be found at the following link:

https://github.com/iNavFlight/inav/releases/latest

The latest release candidate (possibly unstable) can always be found at the flowing link:

https://github.com/iNavFlight/inav/releases

jaimegil118, 2019-02-19:
Thanks, Tim
If that’s the case, then the info from Inav is confuse.
I mean:
Working with Inav 2.0. Each time I open the configuration it says: a new
version is available ....
Then, if you open, you get 2.1.3. It fiesn’t Say it’ a candidate for
release. Appears as a new version
When zonas Switched fro 1.9 to 2.0 , this didn’t happened. It was crtstal /
clear
Regards
Jaime

krzysztofmatula, 2019-02-19:
I think you confuse the configurator version with the firmware version.
Latest configurator is 2.1.3.
Latest firmware is 2.1.0-RC3

Pairan, 2019-02-19:
@jaimegil118 you have to drag the download into your "PROGRAMS" folder and replace the old one OR start the downloaded configurator from the "downloads" folder.

teckel12, 2019-02-19:
@jaimegil118 It sounds like your just downloading and running a new configurator 2.1.3 each time instead of downloading, installing on top of the old version, then running. So issue #1 is upgrading to 2.1.3 of the configurator correctly.

The second issue is that the latest release version of INAV is 2.0.1.  Version 2.1.0 should be released very soon (which is why there's a new version of the configurator). But, if you want to use the release condidates of INAV, you just need to select that in the configurator. By default, the configurator only downloads release versions, not pre-release.

I hope this makes things clear, and to break up your two problems into two different issues that you can solve independently.

jaimegil118, 2019-02-19:
Ok TIM
Seems you mean select “use unstable releases”
Thanks
Jaime

teckel12, 2019-02-20:
@jaimegil118 If unstable is what you're looking for...

closed 2019-02-20 by DzikuVx