iNavFlight/inav #4395: Updating firmware on a FrSky R9 receiver using f.port config and serialpassthrough fails

leeph, 2019-02-19:
## Current Behavior

While 'serialpassthrough' can successfully be used to upgrade a FrSky R9 receiver firmware that is configured in SBUS+Smartport mode, it is currently problematic when the receiver is in F.port configuration.

Issuing the command 'serialpassthrough [n] 57600' and booting the receiver, the FrSky firmware update tool should detect the receiver bootloader and allow you to flash an update file, however this does not work in f.port configuration.  I believe this is because the F.port protocol operates at 115,200 but the FrSky updater tool requires the old 57,600 Smartport speed. I do not think iNav is honouring the port speed change when issuing the serialpassthrough command.

## Steps to Reproduce

1. Flight controller with R9 receiver (R9MM, R9Slim+ etc) configured in F.port mode
2. Issue 'serialpassthrough x 57600' where x = UART number -1
3. Execute FrSky Updater Tool with USB COM port number and firmware file
4. Power receiver - bootloader should be detected by updater tool, but never is.

## Expected behavior

FrSky Updater tool should detect the receiver bootloader and commence updating.

## Suggested solution(s)

I have worked around this problem by tricking inav into thinking the UART which is currently setup for F.Port is actually a Smartport UART, i.e by setting serialrx provider to SBUS and assigning the Smartport telemetry flag for the UART that has the single signal wire from the receiver. This forces inav to set the port speed to 57600 which is a requirement for the firmware update. Serialpassthrough can then be commanded as normal and the receiver is detected and updated.

After flashing, the system is returned to F.port configuration.

So basically iNav is not honouring the port speed change in F.port mode. iNav is running the port at 115,200 which is a requirement for F.port, and not dropping the rate when commanded via the 'serialpassthrough x 57600' command, therefore the updater tool never detects the receiver bootloader which is expecting the 57600 baudrate.

---
My setup is a Matek F411-Wing, FrSky R9MM running inav 2.1 RC3, but this problem affects all versions of iNav that support the 'serialpassthrough' command and F.port modes.

leeph, 2019-04-09:
Bump? This not an issue anyone else is having? Can't just be me, surely!

DzikuVx, 2019-04-09:
I'm not sure this is something for INAV to actually fix. passthrough is a passthrough. Is it working with BF for example?

leeph, 2019-04-09:
Unconfirmed... I suspect the same behaviour applies.

The problem seems to be that iNav does not fully honour the CLI command.

If - on an f.port enabled craft, I issue the command 'serialpassthrough 1 57600' - and UART identifier number 1 is configured for f.port, then it's set for 115,200 (because that's what f.port seems to run at) and iNav does not honour the serialpassthrough command request to drop the port speed to 57600 for the duration of the serialpassthrough process.

If the command would actually force the port speed, I think it would work.

closed 2019-06-22 by stale[bot]

VICLER, 2020-04-26:
I have the same issue. FRsky does not recognizes the receiver. It works fine on betaflight.
I'm using Inav 2.4.0 and configurator v2.4.1.
Kakute f4v2

mattearlypwns, 2020-05-25:
I have been having the same issue