iNavFlight/inav #4401: Problem with Horizon or any stabilization mode

DonaldHillNZ, 2019-02-20:
Hi. Ive been struggling a little with Inav and the setups on a Matek405 wing. Ive pretty much sorted most issues but something is happing I cannot understand or fix. Everything in Manual mode works fine with the plane properly trimmed out. Turning on Horizon mode is fine until I actually try and use the ailerons in this mode. In the air, the plane will suddenly nose down or act very strangely. After researching this on the ground I now notice that whilst the elevator responds correctly to correct a nose down or up in the stabilization mode the minute I touch the ailerons the elevator goes to a down attitude ie. not full down but down. The amount of down corresponds to the amount of aileron I am inputting. There is no mixing at all in the TX and the only mixing is as per the Inav. Im using 2.0.1 Inav. Ive removed any Modes that I setup in case that was causing it but nothing seems to change it. Here is my CLI printout.

# version
# INAV/MATEKF405SE 2.0.1 Oct 31 2018 / 19:28:11 (a04e68aaa)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 100 0
smix 1 4 0 100 0
smix 2 5 2 100 0
smix 3 2 1 100 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 -100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 -100
servo 5 1000 2000 1500 -100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
[... 850 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

shellixyz, 2019-02-27:
You're probably experiencing attitude drift. It can be caused by excessive vibrations and also if you're turning the acceleration is not parallel to the ground anymore and makes the measured attitude drift. It is a known issue. The first thing to do is to check the amount of vibrations and make sure your motor is smooth and that your propeller is balanced

DonaldHillNZ, 2019-02-27:
Hi thank you for your reply. I really am at a dead end. Ive checked the vibrations as suggested and from experience, there is nothing abnormal. When I have the aircraft on the ground and have it in stabilization mode ie. horizon all the control surfaces work as expected ie. tilt the plane down the elevator goes up to recover and visa versa when tilted up. Ailerons work as expected ie. plane tilts left ailerons compensate to bring back to level. But.. the minute I touch the tx and input ailerons to turn one way or the other the elevator moves down in a direct correlation to the amount of aileron I have inputted. Ive checked rechecked and there is no mixing via the tx. Short of pulling out the Matek 405 I cant think of anything else to do. Is there any adjustments deeper down Im unaware of I can try?

DzikuVx, 2019-02-27:
Please verify is channels in radio are not reversed. 
Right stick right should increase ROLL, 
Right stick high should increase PITCH
Left stick right should increase YAW

If not, reverse channels in the radio and then apply rates to servos to match required movements

DonaldHillNZ, 2019-02-27:
Hi Pawel,

I will recheck shortly. I was very careful to make sure they went the 
right way but Ill recheck. I am quite familiar with setting up 
Betaflight with drones so its not new to me but just in case Ill 
recheck.

Best regards

Donald

<author@noreply.github.com>
Sent: 28/02/2019 11:00:51 AM
mode (#4401)

DonaldHillNZ, 2019-02-27:
Hi again Pawel,

I can confirm that right stick aileron in mode 1 to right increases roll 
ie. on the screen goes to 2000
I can confirm that left stick elevator in mode 1 forward increases pitch 
ie. on the screen goes to 2000
I can confirm that left stick rudder in mode 1 to right increases the 
yaw ie. on screen goes to 2000.

Am I correct here. Its how I have always understood it to be?

Throttle in mode 1 forward stick increases again on screen goes to 2000.

Im really puzzled with this problem. I have had quite a few years of 
experience but only started with the Inav six months ago.

Best regards and thank you for the help.

Donald

shellixyz, 2019-02-28:
The next step is to check the outputs on the motor tab. If you can make a short video showing how the output values change with inputs. Then if it is all right make sure you plugged the right servo on the right output. You can see the output mapping in the mixer tab.

DonaldHillNZ, 2019-02-28:
Ok will do. And again thanks for the help

DonaldHillNZ, 2019-02-28:
Hi Michel, I can send you a video if you like but trying to get it smaller, HOWEVER, I have noticed that in the motor tab the outputs show the reverse to what is in the receiver tab. They show low values when pushed to the right or forward in the case of the pitch. Ive never looked in here before. How do I correct this and would this cause my problem?

DonaldHillNZ, 2019-02-28:
The only one that is correct in the motor tab is the yaw. It goes the correct way. Ill wait until I hear from you before making any changes so I don't complicate things

shellixyz, 2019-02-28:
> They show low values when pushed to the right or forward in the case of the pitch

That's because you reversed the servo outputs. Notice the -100 rates from your config:

```
servo 2 1000 2000 1500 -100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 -100
servo 5 1000 2000 1500 -100
```

But that's not your problem. You just need to set this right so that your servos move in the right direction.

The direction it moves is not important for this step. What you need to check is that when you move only the aileron channel the output that's assigned to the elevator doesn't move.

DonaldHillNZ, 2019-02-28:
Hi again Michel,

Ok heres what Ive found.

When the mode is manual everything moves as expected apart from the reverse we discussed.

However when I activate the Horizon mode there is a tiny movement of the elevator downwards or increase in value. This isnt noticeable visibly unless you watch very closely but as I increase the amount of aileron the elevator increases the value ie. more and more down depending on the input of the aileron. This is in the motor tab.

Ive basic modes setup but have attached photos of the modes.

Ive made a short vid of what happens in the motor tab.
[image]
[image]
[image]
[image]

[20190301_091633.zip](https://github.com/iNavFlight/inav/files/2916765/20190301_091633.zip)

.

shellixyz, 2019-02-28:
That's very strange. The servo 3 output gets lower when you move the aileron stick both ways. Even more so because you set the mid for servo 3 to 1500 so when your sticks are centered the servo 3 output should be 1500 and it is not. And again just to be 100% sure when you move the aileron stick the pitch channel doesn't move ? I will try your mixer settings to see if I can reproduce your issue.

shellixyz, 2019-02-28:
What was the modes enabled when you did the test on video ? Did you have the `turn assist` flight mode enabled ?

DonaldHillNZ, 2019-02-28:
Only modes enabled were throttle disarmed and horizon mode enabled.

Sent: 1/03/2019 10:33:54 AM
mode (#4401)

DonaldHillNZ, 2019-02-28:
And yes there is a very slight movement in the elevator as you see when I enable the horizon mode. Its always increase in value. If you look in the Receiver tab there is no movement at all of the elevator however Im about to check that again. Haha getting paranoid now.

DonaldHillNZ, 2019-02-28:
Im here now so will answer you as soon as I get  a message

shellixyz, 2019-02-28:
Ok I know what is going on. Everything is perfectly normal. The servos are not at 1500 even with sticks centered because you're board is not laying flat, my bad. That's normal the servos are trying to correct the attitude. Also with horizon mode the more you move the sticks from center the more you are transitioning from angle mode to acro mode so it tries to correct the pitch less and less the more you move the aileron stick that's why the elevator servo goes back to center (you're not moving the elevator stick). Horizon mode is a mix between acro and angle.

DonaldHillNZ, 2019-03-01:
I can completely confirm that even though the elevator is moving as shown in the vid attached there is no movement shown on the receiver tab but there is on the motor tab as my previous vid. Ive put the plane into and extreme position ie pointing down and as you can see the elevator is in the up trying to correct but when the aileron are moved so does the elevator.
[20190301_105618.zip](https://github.com/iNavFlight/inav/files/2917073/20190301_105618.zip)

DonaldHillNZ, 2019-03-01:
ok does that mean I should be using angle mode for the stab instead of horizon mode?

shellixyz, 2019-03-01:
Since you're new to iNav you should definitely start with angle mode or acro/rate mode (no mode enabled) and leave horizon mode for later when you're more familiarized with the basic flight modes. In any case you should always have manual mode on your mode switch(es) in case the plane is not behaving the way you expect.

You can find more information about the available flight mode here: https://github.com/iNavFlight/inav/wiki/Flight-modes

DonaldHillNZ, 2019-03-01:
Thanks Michel.

I was following the mode settings from ragthenutsoff so ill go back to 
my instincts.

Gold support so thank you

Sent: 1/03/2019 1:52:56 PM
mode (#4401)

DonaldHillNZ, 2019-03-02:
Hi Michel,

Wallah problem solved using Angle mode and not horizon mode. Thank you for all your time. As I mentioned I simply followed ragthenutsoff way of setting up the modes and did look at the info on the wiki. Im now setting very basic modes and going to test fly again.

shellixyz, 2019-03-02:
Glad that I've been able to help and everything is working normally. Happy flying !

closed 2019-03-02 by shellixyz