iNavFlight/inav #4404: F722-SE doesn't beep to indicate stick control such as acc calibrate and save settings

laurentopia, 2019-02-23:
## Current Behavior
F722-SE doesn't beep to indicate stick control such as acc calibrate and save settings

## Steps to Reproduce
do the stick control for acc calibrate or mag calibrate
= doesn't beep
however if I mag calibrate from within inav configurator it beeps

## Expected behavior
the FC beeps to indicate calibration

## Suggested solution(s)
make the FC beep like when it's connected to the computer

---
# INAV/MATEKF722SE 2.1.0 Feb 10 2019 / 16:42:36 (0d3d0ff30)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
dump:
https://pastebin.com/g6X2qt5q

teckel12, 2019-02-23:
@laurentopia My Matek F722SE running 2.1.0RC3 beeps just fine.

Did you have your battery connected?

krzysztofmatula, 2019-02-23:
My F722SE beeps as well (al least while calibrating MAG, didn't check ACC).

laurentopia, 2019-02-24:
Yes with and without battery connected. Only beeps when USB connected.
Do yours beep when on battery only?

teckel12, 2019-02-24:
@laurentopia I'll test right now.

teckel12, 2019-02-24:
@laurentopia Just tried on my F722SE running 2.1.0RC3 and it beeps with or without USB connected. 

Maybe related to #4377 ?

teckel12, 2019-02-24:
@laurentopia Try disabling the feature DASHBOARD

b14ckyy, 2019-02-28:
can you try to do the same command 2 times? I need to repeat the same battery selection command 2 times and only on second times it beeps. Then it could be really related to my issue.

laurentopia, 2019-03-01:
Sorry guys, lost the 722 so I cannot reproduce ;(

closed 2019-05-14 by stale[bot]