iNavFlight/inav #4406: Matek F722-SE fails during Run-time calibration with analog rssi input enabled

ericogr, 2019-02-24:
## Current Behavior

Matek F722-SE fails during Run-time calibration.

## Steps to Reproduce

1. Reset settings (run-time calibration works fine)
2. Enable analog rssi input (run-time calibration does not work anymore)

## Expected behavior
run-time calibration working

## Suggested solution(s)

## Additional context
You can setup, calibrate model with radio transmitter on but it doesn't work anymore.
Aparently values read from rssi are ok.
[PasteBin](https://pastebin.com/rgGtyQ6v)

- INAV/MATEKF722SE 2.1.0 Feb 10 2019 / 16:42:36 (0d3d0ff30)
- GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

digitalentity, 2019-02-25:
Check the latest release please

ericogr, 2019-02-26:
updated to latest version:

- INAV/MATEKF722SE 2.1.0 Feb 25 2019 / 17:01:07 (65b0ec142)
- GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

**I have the same issue :(**

ericogr, 2019-02-26:
[SOLVED]
I found my fault ... in the default configuration of sensors, "pitot tube" comes configured for ADC. I didn't have this sensor and had not noticed it. I set it to none and the calibration was successful.

closed 2019-02-26 by ericogr