iNavFlight/inav #4408: Quad soaring when RTH landing

Missionhou, 2019-02-25:
I cut the mode to HOLD POSITION mode when Quad RTH and descending，suddenly the Quad's throttle shot high, and was not controlled by the RC throttle. The Quad is soaring! The altitude of the plane rose rapidly,Later I through the RC forced disarm, Quad crashed!
I have a log file: 
 https://www.dropbox.com/preview/LOG00001.TXT?role=personal

Firmware: inav_2.1.0_RC3
Frame: QAV350
motor: DJI Mavic Pro motor
ESC: BLheli 4IN1 
flightControl: OmniBus F4v3 

I don't know what the problem is?
Who can help me

[image]

giacomo892, 2019-02-25:
The log file is 1byte long. Nothing in there

krzysztofmatula, 2019-02-25:
The link to the log on dropbox is incorrect. Instead of copy&pasting from browser url, please make sure the file as actually shared and provide the share link.

As for the issue, please check accelerometer logs where the problem happens. It might be that at some point vibrations match with the accelerometer resonance frequency. This should be clearly visible in the log. If that's the case, soft mounting the FC to the frame will help.

Missionhou, 2019-02-25:
LOG FILE:
https://yunpan.360.cn/surl_yFwxgGf5vhC

Missionhou, 2019-02-25:
The file is too big to past directly（45.3M）.  Is there a way to split the file?

stronnag, 2019-02-25:
Use a compressor; gzip, xz etc. xz compresses it to 11M.

Missionhou, 2019-02-25:
[Uploading LOG00001.zip…]()

krzysztofmatula, 2019-02-25:
It's not looking like the accelerometer resonance case...
[image]

Something that I can't explain happens when you go to POSHOLD from RTH at 11:33.
The `navTgtPos[2]` keeps going up. That could indicate that the throttle stick is up, causing the target altitude to increase, but actually `rcData[3]' is quite low, even at rc-command level at the end.
@digitalentity any clue?

digitalentity, 2019-02-25:
In landing phase of RTH we use descent rate control. It's implemented in a such way that navTgtPos[2] is set below navPos[2] in such way that Pos-to-Vel controller will yield the desired descent rate.

I see the only problem here - accelerometer resonance case - should be solvable with better vibration dampening.

krzysztofmatula, 2019-02-25:
The above case is after exiting the RTH and staying in POSHOLD (and navTgtPos[2] quickly goes up).
Looking at the ACC traces they don't look as bad as I once have in my OmnibusF4V3 setup, but...

Missionhou, 2019-02-26:
I will change another type frame to test this problem,Thanks for everybody!

krzysztofmatula, 2019-02-26:
Please make sure the FC is soft-mounted. 
If possible, try soft-mounting on the same frame to see if it is helping.

Hint: if it happens again don't panic and don't disarm. Just disable RTH and POSHOLD and take the machine down manually in ANGLE (or ACRO).

Missionhou, 2019-02-27:
FC is no soft-mounted.
I will try soft-mount.

closed 2019-05-12 by stale[bot]