iNavFlight/inav #4409: Problem throttle in poshold

alfa79, 2019-02-25:
good afternoon, I hope to be writing in the right place since I never do. Let's see if someone can help me. I have the kakute F7 and it goes pretty well, the problem comes when I put the poshold mode, in that way to give a little accelerator the drone is accelerated to the crazy, I can not lift the flight xq would crash. in the other modes it goes smooth. thanks to everyone. a greeting.

teckel12, 2019-02-25:
@alfa79 What version of INAV are you running?  Have a log of this or at least a CLI dump?

teckel12, 2019-02-25:
Related to #4408 ?

alfa79, 2019-02-25:
Hello Teckel12, thanks for replying. 
The version I use is inav 2.1.2.
I look at how the registration is done to send it to you, the CLI dump I'll copy it next.
Thanks for your help.

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/KAKUTEF7 2.1.0 Dec 13 2018 / 16:45:50 (3dc3158ac)
# GCC-7.1.0

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature BAT_PROF_AUTOSWITCH
feature GPS
feature TELEMETRY
feature CURRENT_METER
[... 768 lines cut ...]
set tpa_breakpoint = 1390
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 360
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

digitalentity, 2019-02-25:
I'm not sure what the issue is - description is not clear. If it's uncontrolled climb - it's very likely accelerometer resonance issue.

alfa79, 2019-02-25:
hi digitalentity, in stabilized and althold works well. The drone I can not take off because when giving a little throttle, some propellers are accelerated a lot, preventing that I can take flight because it would crash.

alfa79, 2019-02-26:
hello,
I enclose the data dump that you asked me, first flying a few seconds in althold and then trying to take off in poshold. The two modes of flight in the interior of the house.
thank you.
[LOG00002.TXT](https://github.com/iNavFlight/inav/files/2907389/LOG00002.TXT)

teckel12, 2019-02-26:
@alfa79 When you say take off in POSHOLD, do you mean you're on the ground/floor not armed, you enable POSHOLD, then arm?  I would think activating POSHOLD while on the ground could be bad. 

Also to clarify, are you trying to do POSHOLD indoors? That sounds like a **really** bad idea if I'm understanding you correctly.

alfa79, 2019-02-26:
the steps that I take are the following ... armo motors, I put the stick in poshold mode and then I try to fly it. It is the same procedure that I follow for althold.
I understand from your comment that I must activate poshold once it is flying in althold ??
I do the tests indoors since I have 8/9 satellites.
I understand that I have to change those two questions that you have told me.
I try to tell you the result.
I started the poshold from the ground because when I fly in angle and I pass to althold the drone makes a strong climb.
Thank you for your interest, I'll try it soon and I'll tell you.
a greeting.

teckel12, 2019-02-26:
@alfa79 First off, even if you have multiple satellites indoors, you shouldn't use POSHOLD. GPS location depends on VERY precise timing. When inside, you're getting reflection satellite signals, this can make the model believe it's moving all over the place (even very quickly), and try to compensate. Don't be surprised if it slams into a wall or yourself, you've been warned.

As far as ALTHOLD surging when activated, that's because you don't have your hover throttle set correctly. You should set this so there's no surge. Then, activate ALTHOLD once you're already in the air.

Finally, while ALTHOLD will kinda work indoor, don't expect it to work like a DJI Mavic, which doesn't use a barometer to hover indoors. A barometer may fluxuate by a couple meters, even while stationary. And indoors you probably have less than 3 meters from the floor to the ceiling. 

Basically, don't count on ALTHOLD working well indoors and I wouldn't use POSHOLD at all inside.

DzikuVx, 2019-03-04:
From the log it looks like you have `nav_mc_hover_thr` wrong. Log suggests that it should be around 1300-1350. On top of that, taking off in AltHold-PosHold is stronly not advised

closed 2019-03-04 by DzikuVx

alfa79, 2019-03-04:
Thank you, this week when I can go to the mountain, I try it and I tell you. thanks for the interest

alfa79, 2019-03-05:
Hi, I went out to try the drone and it works fine. I change the mode stick and it does not speed up having changed to 1350. In poshold it starts making circles the drone. I guess it's a compass problem. what can be the problems? Thank you.

shellixyz, 2019-03-05:
Toiletbowling means one or more of these issues:
* Bad mag orientation
* Bad mag calibration
* Bad mag placement (magnetic interference from power wires / motor magnets)
* Defective mag

DzikuVx, 2019-03-05:
https://www.youtube.com/watch?v=FlEm0-pXNf0