iNavFlight/inav #4411: 2.1.0 delayed GPS acknowledgement time

digitalentity, 2019-02-25:
Possibly re-occurance of #4178. 
Details: https://github.com/iNavFlight/inav/pull/4193#issuecomment-463075963

Only Matek SAM-M8N seems to be affected, random Chinese M8N modules don't seem to have any issues

Jetrell, 2019-02-26:
I saw a post on a forum recently about this matter.
Are you talk about the Matek SAM M8Q gps units? Not the M8N as said above?

I ran some tests on my M8Q +5883 and noticed the same thing. It has a delay before the flight controller detects the gps module.
I did notice this with RC3 on start up a couple of times when an OSD message appeared across the screen saying NO GPS DETECTED.
It only lasted maybe thirty seconds then disappeared.

Jetrell, 2019-03-04:
> 
> 
> Possibly re-occurance of #4178.
> Details: [#4193 (comment)](https://github.com/iNavFlight/inav/pull/4193#issuecomment-463075963)
> 
> Only Matek SAM-M8N seems to be affected, random Chinese M8N modules don't seem to have any issues

I have noticed this bug has moments when it takes nearly a minute before the flight controller will talk to the GPS. 
Its even taking this amount of time when the GPS has a satellite fix after the smallest reboot.
My concern is that it would only take a small brownout of the flight controller and the GPS connection would be instantly lost for a time.
My BN-180 and M8N don't have this trouble!

The GPS has a lock and it took 30+ seconds with multiply timeouts. (coordinates are removed)

[image]
 

Are the other people that brought up the problem still working with it?

@digitalentity Do you have a Matek M8Q GPS module to test it with? Are Matek sending you one? It would be in their interests to do so.

digitalentity, 2019-03-04:
I have Matek SAM-M8Q. This is a work in progress

Avegawanderer, 2019-03-16:
Confirm the same behavior with Matek 722SE running iNav 2.1.0 and bn880 gps. It takes random time up to minute to get GPS working

Avegawanderer, 2019-03-18:
It appears to be my mistake for reporting this. Reflashed few times, now the problem is gone. Seems like I have been using one of RC, not the stable 2.1.0.

DzikuVx, 2019-05-24:
I think we can move it to Future. My SAM M8Q seems not be affected for example

Jetrell, 2019-05-26:
> 
> 
> I think we can move it to Future. My SAM M8Q seems not be affected for example

When you say ... Not affected. Do you mean that it doesn't have the timeout delay? Or do you mean that it still works?
Mine still works fine.... After 30+ seconds, before the FC recognizes it.  
This affects two different FC's, with the both SAM M8Q and the Mag version as well.

romko1, 2019-06-05:
@digitalentity I found out where the mistake is. As I wrote above I have a GPS module Radiolink TS100 mini. If I connect to a given GPS module via FTDI cable (u-center) I preset it (transfer rate, frequency ... etc) and save, then connect GPS module to FC MatekSys F722 mini it works fine. However, if the GPS module power is longer off (more than 3 hours) then the GPS module will lose my settings (defaults to 9600 speed, frequency..etc) and will turn on the FC MatekSYS F722 mini - automatic configuration approx. 20 seconds red - buzzer bleeping (HDOP is 0.0 not 99), and then get caught - shining blue (HDOP is 99). Wouldn't it help to set a larger delay in the firmware for automatic configuration? If so, where can I please change it? Thank you.

closed 2022-02-10 by DzikuVx