iNavFlight/inav #4412: Inav 2.0.1 gone arrow dissapers

skinnyguyfpv, 2019-02-25:
I have recently installed a mate f405-std and bn180 GPS everything is fine apart from my home arrow blinking then when I arm I'm left with a small line that blinks so can't tell were home is I've been ripping my hair out, I have a 3d fix and 15-16 sats

krzysztofmatula, 2019-02-25:
Did you upload the current OSD font?

skinnyguyfpv, 2019-02-25:
Yeah Mate, I've installed 3 planes with inav now and have been very successful but this one for some reason is rather hard, I'm thinking it might be a bad gps the reason I say that is it says it has a 3d fix but in inav configurator there is no map, i have played arround with gps settings and diffrent baud rates etc what are your thoughts?

skinnyguyfpv, 2019-02-25:
The arrow Looks he same as a thread #2186 but my arrow is on its own in the screen so no overlapping

shellixyz, 2019-02-27:
Please make a short video of the issue if possible

shellixyz, 2019-02-27:
Also you should now upgrade to 2.1.0. Issues in previous versions won't be fixed.

closed 2019-03-17 by DzikuVx

jm-wijnholds, 2019-05-16:
I have exactly the same problem. When disarmed the home arrow is blinking, when arming it changes to a blinking line. 

I also noticed that I can’t enable RTH and launch. I have enough satellites. 

Some ideas what the problem might be?

teckel12, 2019-05-16:
@jm-wijnholds Number of satellites does not mean you have a GPS lock. You can have 10 satellites but not a lock. The "big myth" is that # sats = GPS lock, which is not at all true.

Try turning on extra arming safety and wait for a GPS lock and then see what happens.

ucakar, 2020-06-20:
Hello,

I have sam problem, is there any solution?

Thanks,

Andrej

wizard220s, 2020-07-01:
I have the same problem with two different F4  ( Matek 405 STD and Omnibus F4 V3) the arrow and the number of  sats has dissapear and RTH do not workn althought all is in blue ( Mag GPS Baro Etc)

wizard220s, 2020-07-01:
I forgot to comment that it happens with INAV 2.5 also

wizard220s, 2020-07-01:
The GPS is BN 880 
I have reviewed all cli without success.
Please help!