iNavFlight/inav #4415: 2.1 release temperature sensor

Jetrell, 2019-02-26:
## Current Behavior
Using configurator 2.1.4 and firmware 2.1 full release.

## Steps to Reproduce
Following the temperature sensor wiki.
 https://github.com/shellixyz/inav/blob/e02d4f7f24cde98052835e8c5a60e0ee3ae02555/docs/Temperature%20sensors.md

## Expected behavior
The ability to display a Label or OSD Symbol in front of the exterior temperature sensor value.

## Additional context
I have entered these setting into the CLI.
[image]

 I can not get the OSD symbol or the Label to be displayed in front of the temperature value.  It just shows 32C.

It does show the IMU and BAR symbol before the flight controller temperature sensors. It only appears to effect the exterior sensors.

I have followed the wiki to the best of my knowledge. 
Is it possible that I have missed something?

shellixyz, 2019-02-26:
You need to update the OSD fonts on your FC. The temperature symbols have just been added in the configurator in version 2.1.4

Jetrell, 2019-02-26:
@shellixyz  You need to update the OSD fonts on your FC.

Hi, I am using the latest 2.1.4 configurator that I assume to have the latest fonts. I have also uploaded a new font into the flight controller. 
It must have those symbols because the IMU and BAR symbol appear on the OSD. While the exterior symbols do not.
In the previous RC3 I could get the temperature label to appear before the temperature value with a large gap between the two. Now I can not get a Label or a symbol to appear.

shellixyz, 2019-02-26:
Strange. I tried to reproduce the issue but it is working fine for me 2.1.0 FW and 2.1.4 configurator both the symbol and label. Which font are you using ?

[image]
[image]

Jetrell, 2019-02-27:
I had not got around to checking the actual OSD image as you have posted.
I was just looking at the preview on the configurators OSD page.

After looking at the actual OSD image, it does display them on the OSD which is the most important thing. _But it doesn't display the label or the symbol on the OSD preview page in the configurator._ I can live with that :+1: 

thanks for your time @shellixyz

shellixyz, 2019-02-27:
Ha, good ! I'm aware about the OSD preview. It is not ideal but it is "normal". I guess I should add a warning in the documentation and I'll probably do something about it for the next release.

Thanks for the feedback

closed 2019-02-27 by shellixyz