iNavFlight/inav #4417: With upgraded firmware when you use mixer sheet in the INAV configurator yu will warn that:  "You need to upgrade your firmware before you can use the Mixer tab."

robiu, 2019-02-27:
## Current Behavior
with autoconnect enabled try to use and save rebooting from a mixer sheet, after the restart you are waarning about to upgrade a firmware even if it's already upgraded and after restart the gui connect and disconnect

## Steps to Reproduce
watch the movie
## Suggested solution(s)
if you decheck AUTOCONNECT, the problem disappear but the gui execute an autoconnection after restart

[issue-mixer-autoconnect.zip](https://github.com/iNavFlight/inav/files/2908967/issue-mixer-autoconnect.zip)

DzikuVx, 2019-02-27:
Do not use Autoconnect on configurator. It's not stable and will be probably removed in next release

closed 2019-02-27 by DzikuVx

robiu, 2019-02-27:
you said what I wrote, the problem is that without the autoconnect active the configurator reconnects even if you should not reconnect

DzikuVx, 2019-02-27:
you are mistaken and you mix Autoconnect (connect when new device was plugged in and Configurator detected a new COM port) with Reconnect after you rebooted the board from one of the tabs. 

Reconnect work now always and there is no need to use Autoconnect which was designed for something completely different and will probably be removed in a future