iNavFlight/inav #4421: Omnibus F4 V6 (not nano) servo setup problem

CRISTSC, 2019-02-27:
## Current Behavior
In the new release, inav_2.1.0_OMNIBUSF4V6.hex, airplane setup not working. Motor 1 and 2 working, the first 2 servos are working but the rest are not assigned to any output.

[image]

## Steps to Reproduce

1. airplane basic setup
2.
3.
4.

## Expected behavior
Have 2 motors output and 4 servo output

## Suggested solution(s)
Add CH5 (servo 3) to C08 and CH6 (servo 4) to C09.
Mr. ASIZON already did this but somehow all 4 servos are working but not the motors.
(https://github.com/iNavFlight/inav/files/2891114/inav_2.1.0_OMNIBUSF4V6.zip)
 I do not know why but i think that supposed to be 3 separate Timers: 1 for first 2 motors, 1 for Ch3 and Ch4 and 1 for Ch5 and Ch6, so the board can be used in both airplane and multi rotor configuration.

# version
# INAV/OMNIBUSF4V6 2.1.0 Feb 25 2019 / 17:06:58 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

DzikuVx, 2019-02-27:
From what I see in the code, V6 is really OmnibusF4 but a variant of Fireworks target. And that means, it has only 2 servo compatible outputs not 4 like previous variants of Omnibus F4

CRISTSC, 2019-02-27:
In the modified code i have all 4 as it is normal but no motor. This board is compatible with V4Pro but different pinout and 6 uarts.

lgojniczek, 2019-02-27:
Also KFC32F3 have similar problem on new Inav 2.1.0

All servo outputs are "dead"

DzikuVx, 2019-02-27:
On the topic of KFC, @marbalon would have to fix it as none of the devs actually have the board.
And no, OmnibusF4V6 is a different problem since there are just too many different versions of hardware with the same name and currently noone knows what goes where

CRISTSC, 2019-02-27:
i have the pin-out for F4V6 if needed.
**A00: RSSI IN**
**A01; SERIAL4 UART RX
 A02: SERIAL2 UART TX**
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
**A08: FREE**
A09: SOFTSERIAL1 UART TX/RX
A10: FREE
A11: USB IN
A12: USB OUT
A13: FREE
A14: FREE
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE
B03: BARO CS
B04: BEEPER OUT
B05: MOTOR4 OUT
**B06: LED1 OUT**
**B07: CAM_C**
B08: I2C1 SCL
B09: I2C1 SDA
B10: SERIAL3 UART TX
B11: SERIAL3 UART RX
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE
C01: ADC CH1
C02: ADC CH2
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL6 UART RX
**C08: MOTOR 5 OUT
 C09: MOTOR 6 OUT**
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SPI_PREINIT CS

CRISTSC, 2019-02-27:
pinout for F4V6:
**A00: RSSI IN**
**A01; SERIAL4 UART RX
 A02: SERIAL2 UART TX**
A03: MOTOR3 OUT
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
**A08: FREE**
A09: SOFTSERIAL1 UART TX/RX
A10: FREE
A11: USB IN
A12: USB OUT
A13: FREE
A14: FREE
A15: OSD CS
B00: MOTOR1 OUT
B01: MOTOR2 OUT
B02: FREE
B03: BARO CS
B04: BEEPER OUT
B05: MOTOR4 OUT
**B06: LED1 OUT**
**B07: CAM_C**
B08: I2C1 SCL
B09: I2C1 SDA
B10: SERIAL3 UART TX
B11: SERIAL3 UART RX
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE
C01: ADC CH1
C02: ADC CH2
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: SERIAL6 UART RX
**C08: MOTOR 5 OUT
 C09: MOTOR 6 OUT**
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SPI_PREINIT CS

DzikuVx, 2019-02-27:
@Asizon I think this is for you since you implemented original V6 target

CRISTSC, 2019-02-27:
@Asizon already has this pinout and modified the code to work with 4 servo but unfortunately no motor output yet.

Asizon, 2019-02-27:
This target needs to define 2 new timers to get what you are looking for, unfortunately I do not understand very well that 2 different timers are assigned per motor, I do not understand how airplane works, sorry

CRISTSC, 2019-02-27:
can i try to find the solution starting from what you already modified for the last hex you sent me? i have the board and enough time to try until i succeed or destroy the board, but at least i know i tried everything. please

Asizon, 2019-02-27:
Yes of course, these are defined Timers:

    // Motor output 1: use different set of timers for MC and FW
    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MC_MOTOR,                       1, 0), // S1_OUT    D(1,7)
    DEF_TIM(TIM8, CH1, PB0,                    TIM_USE_FW_MOTOR,   1, 1), // S1_OUT    D(2,2,0),D(2,3,7)

    // Motor output 2: use different set of timers for MC and FW
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MC_MOTOR,                       1, 0), // S2_OUT    D(1,2)
    DEF_TIM(TIM8, CH2, PB1,                    TIM_USE_FW_MOTOR,   1, 1), // S2_OUT    D(2,2,0),D(2,4,7)

    DEF_TIM(TIM2, CH4, PA3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    1, 1), // S3_OUT    D(1,6)
    DEF_TIM(TIM3, CH2, PB5, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    1, 0), // S4_OUT    D(1,5)
	DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0), // S5_OUT
    DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0), // S6_OUT

Asizon, 2019-02-27:
these are defined timers:
    // Motor output 1: use different set of timers for MC and FW
    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MC_MOTOR,                       1, 0), // S1_OUT    D(1,7)
    DEF_TIM(TIM8, CH2N, PB0,                    TIM_USE_FW_MOTOR,   1, 1), // S1_OUT    D(2,2,0),D(2,3,7)

    // Motor output 2: use different set of timers for MC and FW
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MC_MOTOR,                       1, 0), // S2_OUT    D(1,2)
    DEF_TIM(TIM8, CH3N, PB1,                    TIM_USE_FW_MOTOR,   1, 1), // S2_OUT    D(2,2,0),D(2,4,7)

    DEF_TIM(TIM2, CH4, PA3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    1, 1), // S3_OUT    D(1,6)
    DEF_TIM(TIM3, CH2, PB5, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    1, 0), // S4_OUT    D(1,5)
	DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0), // S5
    DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0), // S6

CRISTSC, 2019-02-28:
fixed for bench test. tested with airplane and quad. thanks for help @Asizon 
[inav_2.1.0_OMNIBUSF4V6.zip](https://github.com/iNavFlight/inav/files/2916664/inav_2.1.0_OMNIBUSF4V6.zip)

target.c
const timerHardware_t timerHardware[] = {
    DEF_TIM(TIM10, CH1, PB8, TIM_USE_PPM,                           0, 0), // PPM

    // Motor output 1: use different set of timers for MC and FW
    DEF_TIM(TIM3, CH3, PB0, TIM_USE_MC_MOTOR,                       1, 0), // S1_OUT    D(1,7)
    DEF_TIM(TIM8, CH2N, PB0,                    TIM_USE_FW_MOTOR,   1, 1), // S1_OUT    D(2,2,0),D(2,3,7)

    // Motor output 2: use different set of timers for MC and FW
    DEF_TIM(TIM3, CH4, PB1, TIM_USE_MC_MOTOR,                       1, 0), // S2_OUT    D(1,2)
    DEF_TIM(TIM8, CH3N, PB1,                    TIM_USE_FW_MOTOR,   1, 1), // S2_OUT    D(2,2,0),D(2,4,7)

    DEF_TIM(TIM2, CH4, PA3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    1, 1), // S3_OUT    D(1,6)
    DEF_TIM(TIM3, CH2, PB5, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    1, 0), // S4_OUT    D(1,5)

    DEF_TIM(TIM3, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0), // S5_OUT    D(1,7)

    DEF_TIM(TIM3, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,    0, 0), // S6_OUT    D(1,8)

    DEF_TIM(TIM4, CH1, PB6, TIM_USE_LED,                            0, 0), // LED strip D(1,0)

    DEF_TIM(TIM1, CH2, PA9, TIM_USE_ANY,                            0, 0), // SS1
};

target.h
// Number of available PWM outputs
#define MAX_PWM_OUTPUT_PORTS    8
#define TARGET_MOTOR_COUNT      6

Asizon, 2019-03-02:
Im happy for your solution, you changed s5 ,S6 TIM8 to TIM3, right? Are you try the firmware and works correctly?

CRISTSC, 2019-03-02:
yes. TIM8 was designed for FW Motor and even if motor is set to 50Hz is in conflict with servo. i think that TIM2 is for tricopter servo. so far everything is working correctly. because is the first time using Inav it takes longer to setup and fly but  i am confident that it will fly

Asizon, 2019-03-02:
perfect, do you want me to prepare a pull request to implement it or do you entrust it to yourself?

CRISTSC, 2019-03-02:
please do. is your work. i just modified to satisfy my needs.

closed 2019-03-05 by CRISTSC