iNavFlight/inav #4426: Softserial - camera control (RunCam) in MATEKF405 vs. MATEKF722mini

romko1, 2019-02-28:
Hello,
I can not get control of the camera on the new MATEKF722 mini for FW v.2.1.0, and I did not have to modify anything on target MATEKF405 \ target.c and target.h on the original MatekF405 for FW 2.0.1 and I can control the camera (RunCam).
I downloaded this version: https://github.com/iNavFlight/inav/releases/download/2.0.1/inav_2.0.1_MATEKF405.hex
Of course first I downloaded the original version for MATEKF722mini: https://github.com/iNavFlight/inav/releases/download/2.1.0-RC3/inav_2.1.0_MATEKF722MINI.hex
However, camera control does not work.
So I modified target.c and target.h for MATEKF722mini by MATEKF405 but I can not control the camera unfortunately.
For Target.h I added (MATEKF722mini):
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN      PA0 //tc4 - camera control
#define SOFTSERIAL_1_RX_PIN      PA1 //rx4 - camera control
For Target.c I addes (MATEKF722mini):
DEF_TIM(TIM5, CH1, PA0,  TIM_USE_PWM, 0, 0),                            // TX4 & softserial1
DEF_TIM(TIM5, CH2, PA1,  TIM_USE_PWM, 0, 0),                            // RX4 & softserial1

In the original FW for MATEKF405 this is how camera control is set and operating:
Target.c:
{ TIM5, IO_TAG(PA0),    TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_PWM },  //TX4 softserial1_tx
    { TIM5, IO_TAG(PA1),    TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_PWM },  //RX4 softserial1_rx
Target.h:
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_RX_PIN      PA1  //RX4
#define SOFTSERIAL_1_TX_PIN      PA0  //TX4

#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_RX_PIN      PA2  //TX2
#define SOFTSERIAL_2_TX_PIN      PA2  //TX2

Can you please write me where I'm wrong, why do not I control the camera for MATEKF722mini even after modifying FW?

Thank you.
Roman

shellixyz, 2019-02-28:
You don't need softserial on F7. Softserial is only useful in two cases: when you don't have enough hardware UARTs and when you have a MCU that doesn't have internal UART inverters. I don't know why you can't control your camera but using softserial instead of hardware UARTs won't help.

romko1, 2019-02-28:
Yes, F7 has a lot of UARTs, I did the test and I connected the camera control (TX/RX) to UART1 (recommended for GPS), unsuccessfully :( ...I am desperate...

romko1, 2019-02-28:
Yes, yes yes... I uploaded the FW again to MATEKF722mini, the camera connected to UART4 and the remote control works. Solution? re-recorded standard FW (https://github.com/iNavFlight/inav/releases/download/2.1.0-RC3/inav_2.1.0_MATEKF722MINI.hex) into the board and reset camera.

closed 2019-02-28 by romko1