iNavFlight/inav #4427: Bug in FW or wrong flight controller?

romko1, 2019-02-28:
Hello,
Right after uploading FW 2.1.0 for MATEKF722mini and when I plug in an inav configurator I2C error 45 reports.
In CLI console - resource: I see flawed rows:
C08: G÷?ÊRúáL¾ ø|ïhÌ®¡Ø¿?¯?¿»óS¸eãHïÒÜm¼´Üæbz0àå?n¥ö¥n¢1?Ç?i`4?»Ö%þ?Ãñ?_?÷uÖ?õ?'H ¶oÞng?Q`?üTßCìä?MºrÝqöú6òôOjSiriO^,·o©+??UKLg/.+ªÿ¾Ó?òs¡mDjyónÍ²Ö½k	C1Ù;ãÿì«"V0ýÅW5ÆEÇmS?Ý±}cØæËÆ Zx ÏÞé×1 OUT
C09: G÷?ÊRúáL¾ ø|ïhÌ®¡Ø¿?¯?¿»óS¸eãHïÒÜm¼´Üæbz0àå?n¥ö¥n¢1?Ç?i`4?»Ö%þ?Ãñ?_?÷uÖ?õ?'H ¶oÞng?Q`?üTßCìä?MºrÝqöú6òôOjSiriO^,·o©+??UKLg/.+ªÿ¾Ó?òs¡mDjyónÍ²Ö½k	C1Ù;ãÿì«"V0ýÅW5ÆEÇmS?Ý±}cØæËÆ Zx ÏÞé×2 OUT

Is this a FW error or is the board - flight controller itself wrong?
Thank you.
Roman

shellixyz, 2019-02-28:
I can't reproduce the issue. It is working fine for me configurator version 2.1.4 and FW 2.1.0 final. Make sure you flashed with "full chip erase" selected and you're using the right baud rate. The default is 115200.

romko1, 2019-02-28:
Thank you, I'll take the recommended steps.

romko1, 2019-02-28:
Unfortunately, FW inav also keeps telling me that the F7 MATEKF22mini board has I2C 45 errors.... I also noticed that the memory is only 512kB....

romko1, 2019-02-28:
2019-02-28 @ 20:33:13 -- Detected device with total flash size 512 kiB
2019-02-28 @ 20:33:13 -- Read protection not active
2019-02-28 @ 20:33:21 -- Erased 512 kB of flash successfully
2019-02-28 @ 20:33:41 -- USB device successfully closed

b14ckyy, 2019-02-28:
> Unfortunately, FW inav also keeps telling me that the F7 MATEKF22mini board has I2C 45 errors.... I also noticed that the memory is only 512kB....

That's correct. That is the firmware flash size. Not the Blackbox flash. 

The output errors come from a wrong baud rate setting. Get the same issue when connecting with the wrong baud rate, if i forget to switch in configurator after using Bluetooth on 57600

romko1, 2019-02-28:
I've made the recommended modifications, but I'm unsuccessful. I attach a few print-screens.
[image]
[image]
[image]

shellixyz, 2019-02-28:
Sorry I didn't get that you were talking about the `resource` command. Confirmed there is a software issue there.

shellixyz, 2019-02-28:
Will be fixed in the next release. Thank you for reporting. C8 and C9 are the pinIO pins.

romko1, 2019-03-01:
OK. But my question is: C08 and C09 have been corroborated, but related to the I2C errors 45?
In the CLI console: status:
[image]

I did a little test. I uploaded my FC MATEKF722 mini -> FW Betaflight 3.5.6 and in the CLI console (betaflight it allows) eliminated the resource for the I2C1 B08 et B09 and the I2C error disappeared ... maybe it looks to the board error, or? Thank you.

shellixyz, 2019-03-01:
I don't know, it seems unrelated to me.

shellixyz, 2019-03-01:
If it doesn't increase after the FC is powered up it is nothing to worry about

PR #4431 (2019-03-01): Fix missing owner names
Fixes #4427
commit 539e0f8b (2019-03-01): Fix missing owner names

Fixes #4427
closed 2019-03-01 by shellixyz