iNavFlight/inav #4429: Disarm Delay issue

b14ckyy, 2019-02-28:
## Current Behavior
Randomly, especially after short or long flight times, the disarm is delayed. Sometimes I counted about 1.5s. 

## Steps to Reproduce
1. Only tested on plane
2. Arm with Autolaunch always enabled and Airmode after launch
3. Start plane (doesn't matter if auto launched or manual)
4. fly around or test on bench
5. disarm and check time - randomly it is much longer than 250ms
6. also happened after failed auto-launch and motor was turning in the dirt for about 2s until finally disarmed

## Expected behavior
Always disarm after disarm delay - Is on default 250ms for me. 

## Suggested solution(s)
fix to get correct delay all the time.

## Additional context
My dump: 
[MATEKF405SE 2.1.0 - AR Wing - Final.txt](https://github.com/iNavFlight/inav/files/2916548/MATEKF405SE.2.1.0.-.AR.Wing.-.Final.txt)

Video (second disarm shows issue)
https://www.youtube.com/watch?v=65vTvN1DKvI

Logs from the video:
[LOG00086.zip](https://github.com/iNavFlight/inav/files/2916559/LOG00086.zip)

---
# INAV/MATEKF405SE 2.1.0 Feb 25 2019 / 17:02:18 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

giacomo892, 2019-02-28:
Duplicate.
See https://github.com/iNavFlight/inav/issues/3842
And proposed solution https://github.com/iNavFlight/inav/pull/4250

Closing

closed 2019-02-28 by giacomo892

b14ckyy, 2019-02-28:
oh sorry, didn't see that. thx