iNavFlight/inav #4432: OLED display broken in Inav2.1 for SpracingF3 Board

leocopter, 2019-03-01:
## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

leocopter, 2019-03-01:
Hi. The OLED display is blank in INAV2.1 on the SpracingF3 board. It worked perfectly till INAV2.0.
I tested it on several boards and displays.

wx4cb, 2019-03-01:
as per the release notes:

INAV 2.1 is the last release of INAV that fully supports F3 and especially Omnibus F3 boards. Due to the lack of flash and RAM, future versions of INAV will either remove functions on F3 or not support those boards at all. STM32F3 are strongly discouraged for new builds and existing builds should be upgraded to F7 boards

as far as I know, oled support was removed from some of the F3 boards. I would suggest either compiling your own version or upgrading to at least an F4 board

leocopter, 2019-03-01:
Thanks for your answer. I read correctly that 2.1 was the last version to
fully support F3 boards.
I can compile my own but I can't find what to add to target.h to enable the
Oled display.
Any suggestions ?
Thanks.
Chris

wx4cb, 2019-03-01:
it was removed to make RAM/Flash space for other things. as for what to add i'm not certain. I would have to defer to @teckel12 or @digitalentity for that one

shellixyz, 2019-03-01:
@leocopter Add this to `target.h` to enable it again:

```
#define USE_DASHBOARD
#define USE_OLED_UG2864
#define DASHBOARD_ARMED_BITMAP // skip this to save space
```

You might need to disable other stuff like the GPS/RX/telemetry protocols you're not using in `target/common.h` for the firmware to fit in flash.

leocopter, 2019-03-01:
Eureka ! shellixyz. you're the man ! 
I added : 
#define USE_DASHBOARD
#define USE_OLED_UG2864
#define DASHBOARD_ARMED_BITMAP
and I removed the Rangefinder stuff to save some space : the Oled is back.
Thanks !

shellixyz, 2019-03-01:
Awesome :slightly_smiling_face:

closed 2019-03-01 by shellixyz