iNavFlight/inav #4433: Matek F722-STD i2c errors & max load >100%

OloTka, 2019-03-01:
## Current Behavior
My config is  Matek F722-STD powered by Matek FCHUB-6S. After connecting the board to Inav configurator (first powered by battery) I'm getting i2c errors of 45 and in tasks lists max load is exceeding 100% for battery task. 

Here is dump of CLI for my board:
# version
# INAV/MATEKF722 2.1.0 Feb 25 2019 / 16:59:54 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# status
System Uptime: 65 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 16.88V (4S battery - OK)
CPU Clock=216MHz, GYRO=MPU6500, ACC=MPU6500, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x2, 15126528kB, 06/2016, v2.4, 'SA16G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x20010000, Heap available: 1664
I2C Errors: 45, config size: 4671, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 6, cycle time: 509, PID rate: 1964, RX rate: 49, System rate: 9
Arming disabled flags: ANGLE NAV RX CLI

# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9    3329       4    3.4%    0.5%     24374
 1 -     GYRO/PID    1960     160     132   31.8%   26.3%     28619
 2 -           RX      49      53      41    0.7%    0.7%       237
 3 -       SERIAL     100     687       5    7.3%    0.5%       110
 4 -      BATTERY      49   40053      17  196.7%    0.5%       134
 5 -  TEMPERATURE      99      12       3    0.6%    0.5%        28
 6 -       BEEPER      99      10       4    0.5%    0.5%        48
 7 -          GPS      49     260      26    1.7%    0.6%       140
 8 -      COMPASS       9     226     219    0.7%    0.6%       245
 9 -         BARO      43     241     159    1.5%    1.1%       765
13 -    TELEMETRY     498      17       4    1.3%    0.6%       269
16 -          OSD     248    1053      44   26.6%    1.5%      1209
17 -          CMS      49       5       2    0.5%    0.5%        10
20 -      VTXCTRL       4       4       1    0.5%    0.5%         0
Task check function             6       2                        11
Total (excluding SERIAL)                   266.5%   34.4%

# bootlog
Time Evt            Description  Parameters
   3:  0          CONFIG_LOADED 
   3:  1       SYSTEM_INIT_DONE 
 259: 19   TIMER_CHANNEL_MAPPED  (1, 1, 0, 2)
 259: 19   TIMER_CHANNEL_MAPPED  (2, 2, 0, 2)
 259: 19   TIMER_CHANNEL_MAPPED  (3, 3, 0, 2)
 259: 19   TIMER_CHANNEL_MAPPED  (4, 4, 0, 2)
 259: 19   TIMER_CHANNEL_MAPPED  (5, 5, 0, 2)
 259: 19   TIMER_CHANNEL_MAPPED  (6, 6, 0, 2)
 259:  2          PWM_INIT_DONE 
 759: 21  TEMP_SENSOR_DETECTION 
 960:  9         GYRO_DETECTION  (7, 0, 0, 0)
1451: 10          ACC_DETECTION  (8, 0, 0, 0)
1552: 11         BARO_DETECTION  (4, 0, 0, 0)
1552: 20        PITOT_DETECTION  (0, 0, 0, 0)
1562: 12          MAG_DETECTION  (2, 0, 0, 0)
1673: 21  TEMP_SENSOR_DETECTION 
2080: 13  RANGEFINDER_DETECTION  (0, 0, 0, 0)
2080:  4       SENSOR_INIT_DONE 
2633:  5          GPS_INIT_DONE 
2633:  7    TELEMETRY_INIT_DONE 
3634:  8           SYSTEM_READY 

# resource
IO:
----------------------
A00: SERIAL4 UART TX
A01: SERIAL4 UART RX
A02: FREE 
A03: SERIAL2 UART RX
A04: FREE 
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR6 OUT
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: LED2 OUT
A15: FREE 
B00: FREE 
B01: MOTOR5 OUT
B02: FREE 
B03: SPI3 SCK
B04: SPI3 MISO
B05: SPI3 MOSI
B06: I2C1 SCL
B07: I2C1 SDA
B08: FREE 
B09: LED1 OUT
B10: OSD CS
B11: FREE 
B12: FREE 
B13: SPI2 SCK
B14: SPI2 MISO
B15: 
SPI2 MOSI
C00: ADC CH1
C01: SDCARD CS
C02: MPU CS
C03: MPU EXTI
C04: ADC CH2
C05: FREE 
C06: MOTOR1 OUT
C07: MOTOR2 OUT
C08: MOTOR3 OUT
C09: MOTOR4 OUT
C10: FREE 
C11: FREE 
C12: FREE 
C13: BEEPER OUT
C14: FREE 
C15: FREE 
D02: FREE 

Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# INAV/MATEKF722 2.1.0 Feb 25 2019 / 16:59:54 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -AIRMODE
feature THR_VBAT_COMP
feature BAT_PROF_AUTOSWITCH
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 3 2 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 1 0 900 1250
aux 1 2 0 1300 1700
aux 2 26 0 1300 2100
aux 3 4 3 925 1250
aux 4 3 2 900 1700
aux 5 9 2 900 1275
aux 6 8 4 1700 2100
aux 7 19 1 900 1300
aux 8 21 4 900 1300

# adjrange

# rxrange

# temp_sensor

# osd_layout
osd_layout 0 0 23 1 V
osd_layout 0 1 1 12 V
osd_layout 0 2 0 0 V
osd_layout 0 4 8 6 H
osd_layout 0 11 1 10 V
osd_layout 0 12 1 11 V
osd_layout 0 13 23 6 V
osd_layout 0 14 1 1 V
osd_layout 0 15 23 5 V
osd_layout 0 30 2 13 V
osd_layout 0 34 10 1 V
osd_layout 0 44 0 0 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6500
set acczero_x = 19
set acczero_y = 2
set acczero_z = 11
set accgain_x = 4085
set accgain_y = 4083
set accgain_z = 4100
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = 46
set magzero_y = -171
set magzero_z = 91
set baro_hardware = BMP280
set pitot_hardware = NONE
set rssi_channel = 12
set serialrx_provider = SUMD
set sdcard_detect_inverted = ON
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set current_meter_scale = 179
set model_preview_type = 3
set gps_provider = UBLOX7
set gps_sbas_mode = EGNOS
set inav_reset_altitude = EACH_ARM
set inav_w_z_baro_p =  0.350
set nav_rth_altitude = 3000

# profile
profile 1

set mc_p_pitch = 43
set mc_i_pitch = 40
set mc_d_pitch = 20
set mc_p_roll = 43
set mc_i_roll = 40
set mc_d_roll = 20
set mc_p_yaw = 70
set dterm_lpf_hz = 80
set dterm_notch_hz = 200
set dterm_notch_cutoff = 100
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45

# battery_profile
battery_profile 1

set battery_capacity = 2200

OloTka, 2019-04-02:
I've downgraded firmware to 1.9.0 and there were no I2C errors, moving to 2.0.0 I got 22 errors- that only when airspeed sensor initialize at first start, after saving config, no I2C errors. Version 2.2.0 I got the same 45 errors. Could that be caused by temperature sensor initialization - that was not present in 1.9.0 relaye?

shellixyz, 2019-04-02:
Yes. The more stuff it looks for the more errors you're going to get. Getting errors is not a problem what is a problem is if the counter goes up after boot-up.

OloTka, 2019-04-02:
OK, just undefined the temperature sensor in target.h and compiled new firmware. I2c errors gone now. Happy days.  So, the issues were not identifiable sensors - Airspeed and Temperature, which caused the i2c errors at start of the board. Still got an issue with high max usage of Battery task - should that be something to worry about?

digitalentity, 2019-04-05:
To re-iterate what @shellixyz said before: 
Non-zero i2c errors after boot are not a problem.
Consistently increasing number of i2c errors after boot is a problem.

Nihilistic, 2019-04-05:
@digitalentity What was confusing for me about this, was I used to get i2c error:20 when I was had the -20 foot altitude on OSD issue on my F722-SE. And now with 2.1 full, I get i2c error:45, but the altitude is correct (with the usual typical precision of a BMP280 anyway). This made me think the i2c error displayed was a code, and not an error count. 

The i2c error we're seeing is the number of errors that occurred during the boot up then?

digitalentity, 2019-04-05:
Yes, it's a number of times when I2C driver was unable to communicate with a device. Seeing a small number of I2C errors after the boot up is normal.

closed 2019-04-24 by DzikuVx

SherlockMei, 2019-07-14:
I get I2C errors:250 after boot. My Baro and Compass and GPS are ex, Baro and Compass dosen't work, but GPS can. DALRC F405 AIO, M8N, BMP280. I don't know howto?

teckel12, 2019-07-14:
@SherlockMei This is not an error code, it's the number of i2c errors received. A small number (like 20) is normal. A high number means you have a physical hardware issue (not a problem with INAV) or your using the wrong hardware selection.