iNavFlight/inav #4434: Quad drops a lot when switching from Horizon to Pos+Alt hold

laurentopia, 2019-03-01:
## Current Behavior
Quad drops by nearly 3m when switching

## Steps to Reproduce
fly, up high preferably, then slick to altposhold

## Expected behavior
no dropping

## Suggested solution(s)
ideally fix that, or if it's a tuning issue, add steps to help us fix that

## Additional context
https://pastebin.com/m4zrVetn
matek f405-ctr, aokway dr2205-2300kv, 5065 props

Jetrell, 2019-03-02:
This might help sort it.

PIDs for altitude hold:
1. ALT P - defines how fast copter will attempt to compensate for altitude error (converts alt error to desired climb rate)
2. ALT I - defines how fast copter will accelerate to reach desired climb rate
3. VEL P - defines how much throttle copter will add to achieve desired acceleration
4. VEL I - controls compensation for hover throttle (and vertical air movement, thermals). This can essentially be zero if hover throttle is precisely 1500us. Too much "VEL I" will lead to vertical oscillations, too low "VEL I" will cause drops or jumps when ALTHOLD is switched on.
4. VEL D - acts as a dampener for VEL P and VEL I, will slower the response and reduce oscillations from too high VEL P and VEL I

Inability to maintain altitude can be caused by a number of reasons:
1. insufficient ALT_P and/or ALT_I
2. non-functional baro (please go to "Sensors" tab in Configurator and verify that baro graph changes as you move the quad up and down
3. seriously under-powerd quad (ALTHOLD is able to compensate only to some degree. If your quad hovers at 1700 linear throttle without any expo, ALTHOLD might fail to compensate)
4. Gaining altitude during fast flight is likely due to increased air pressure and that is treated as going down in altitude - try covering your baro with (more) foam.

giacomo892, 2019-03-02:
Increase hover_throttle a bit

laurentopia, 2019-03-03:
Thank you, I'll tune all that more to get a smoother transition.

The quad is maintaining altitude a few seconds after dropping and banking and wobbling about, when this dance is over it's quite stable, unless I approach it with my RX, which seems to freak it out and have it move about.

Why is hover throttle even important if PID for althold handle motors to dynamically match altitude?

laurentopia, 2019-03-03:
Here is a video of what's happening

https://youtu.be/TzQ4565hjeY

DzikuVx, 2019-03-04:
`hover_throttle` has to be close to actual throttle when hovering. If not, PID to altitude needs time compensate. I saw motors were pretty high when you switched to ALTHOLD. 
How about sharing a log to see what is happening internally?

DzikuVx, 2019-03-04:
OK, I'm closing this one since it indeed looks like hover_throttle problem

closed 2019-03-04 by DzikuVx