iNavFlight/inav #4437: Mc-braking fail. Quad shoots forward instead of braking

littlbox, 2019-03-02:
Ok I tested mc-braking today and there was still this rubber band when stopping. Then I increased the max.cruise speed from 500cm/s to 700cm/s and then mc-braking seemed to work almost ; somewhat, it seemed to act a little bit soft and was "rubbery" at the end of the stop . So I upped the cruise speed to 800cm/s. Did a couple of stops but the copter seemed to act funny, gaining altitude after each stop. Last try I really pushed it to gain max speed. The craft never came to a stop but instead shoot forward with no control , until I disengaged pos.hold.

littlbox, 2019-03-02:
Inav 2.1.0

SaschaIoT, 2019-03-03:
I use MC BRAKING with optical flow sensor CXOF (not GPS). In bad light conditions (opflow get bad readings) I have the same problem, that the drone shoot forward instead of stopping. I have to to put the stick in the opposite direction to get the drone to stop. In case of bad GPS or bad optical flow sensor readings the drone should go in neutral position instead of flying forward. But I can not say if drone really shoot forward or it is only not stopping. In good light conditions MC BRAKING works really great for me.

KenImhof, 2019-04-03:
I am having the same problem with 2.1 on my quad.  With a max cruise speed of 1500, the quad continues to launch forward when I release the stick.  Pulling back on the stick eventually gets the quad to stop.  I changed the max cruise speed to 750 and the problem only occurs several times each flight.  I suspect that further reducing the cruise speed will completely fix the issue, but the quad will be so slow it will not be fun to fly in this mode.  I eventually disabled braking and everything is back to normal.

littlbox, 2019-04-04:
That's what I did. Disabled the mode , it's too dangerous. https://youtu.be/ued8LG_NSGQ

teckel12, 2019-04-04:
B

giacomo892, 2019-06-17:
@shellixyz didn't you have this issue once?

shellixyz, 2019-06-17:
I did indeed have this issue when testing the dev branch recently. I'll try to find the logs

sam2b, 2019-07-02:
On 2.2.0, I too had this same problem with braking after increasing the navigation speed to 900cm/s.

faceoff717, 2021-07-07:
The issue is still active in INAV 3.0.1

rts18, 2022-01-06:
@DzikuVx 
I understand that time to fix this feature is difficult to find.  But the fact that nothing has be done after 2 years, shows its not a priority.  
Can I now request that the feature be removed, it would help free up more flash for the F411 and F722.
Or can a warning be issued under the releases? **Its dangerous and a trap to anyone unaware of it.**
#6777

rts18, 2022-01-06:
I understand that time to fix this feature is difficult to find.  But the fact that nothing has be done after 2 years, shows its not a priority.  
Can I now request that the feature be removed, it would help free up more flash for the F411 and F722.
Or can a warning be issued under the releases? **This feature can be dangerous and a trap to anyone unaware of its short comings.**
#6777

closed 2023-10-27 by DzikuVx