iNavFlight/inav #4438: Failure to read parameter settings when pasting to CLI

potater1, 2019-03-02:
## Current Behavior

When setting a parameter list in the configurator, a chunk of the input is skipped in some cases.

## Steps to Reproduce
Paste this batch of parameters in the CLI:
```
set nav_fw_launch_thr = 1750
set nav_fw_bank_angle = 30
set fw_ff_roll = 75
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set roll_rate = 30
set nav_fw_cruise_thr = 1500
set nav_fw_launch_thr = 1750
set nav_fw_bank_angle = 30
set fw_ff_roll = 75
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
```
First 3 parameters are set, the rest are skipped and it says:
```
# n_pit = 600
Unknown command, try 'help'
```
Parameters added to the end of the list are again recognized.

## Additional context

- This bug does not occur in 2.0.0, or even in commit 958b49f so it's quite recent
- Occurs when using Configurator 2.1.3
- In some cases the occurrence of this error isn't deterministic, but you get the exact same error most of the time. The number of lines dropped can vary by 1.
- Serial baud rate doesn't affect this bug.
- Inserting 2 or more consecutive empty lines in the part of the list that gets skipped seems to help: every line is read correctly.

---

- FC Board name and vendor:
SPRacing F3
- INAV version string:
INAV/SPRACINGF3 2.1.0 Mar  2 2019 / 18:54:45 (65b0ec142)
GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

potater1, 2019-03-03:
Doesn't happen with Omnibus F4,

# INAV/OMNIBUSF4 2.1.0 Mar  3 2019 / 19:52:00 (be30c4dd7)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

shellixyz, 2019-03-03:
Please try to reproduce with configurator version 2.1.4

hali9, 2019-03-04:
I reproduce  this in inav-conf 2.1.4

DzikuVx, 2019-03-11:
looks like we have to make Configurator slower again.

shellixyz, 2019-03-11:
This sucks it is already pretty slow

DzikuVx, 2019-03-11:
The option is to make it slower only for the boards without VCP as it looks like only they are affected

giacomo892, 2019-03-11:
F3 boards are also very slow to connect to, it takes quite a bit when pressing connect on the configurator.

shellixyz, 2019-03-11:
@giacomo892 I didn't notice the issue on the Omnibus F3 board I'm doing testing/dev on

giacomo892, 2019-03-11:
@shellixyz odd, I use OMNIBUS too and it is slow to load

shellixyz, 2019-03-11:
Definitely not a problem on Linux. Connecting to Omnibus F3 on Linux: https://youtu.be/0HQwSPDWxjs

closed 2019-05-24 by stale[bot]