iNavFlight/inav #4439: invert yaw & pitch & roll on a switch.

Teraformed, 2019-03-02:
I don't know if this is already possibe.

Considering more and more F7 are coming out with 2 cams input where you can toggle between both with a switch. We'll will see more and more quads consisting of front and back camera.

I just build a quad with extra camera facing backwards. Obviously the stick input won't behave the same way like when you are viewing the camera facing front.

I think this feature would be nice to have.

It should be possible to invert yaw, pitch & roll all at same time when toggling to the second camera facing backwards.

Teraformed, 2019-03-02:
wrong thread.

teckel12, 2019-03-03:
@Teraformed Wait. So this request is so you can fly FPV with the camera facing backwards? Do you change the steering wheel in your car when backing up? Seems that you just need to learn to fly backwards (like baking up in your car). But I do like the idea of flipping everything (although not sure if that would make it more difficult to fly).

shellixyz, 2019-03-10:
I guess the idea is to be able to fly with the rear camera like if it was the front. Like this: https://www.youtube.com/watch?v=fr-3AfwhGGw

BTW only roll and pitch are inverted. Yaw stays the same.

teckel12, 2019-03-10:
I would think you would just get used to the controls flying backwards like when driving in reverse in a car.  But still, can't you just change the mix on the transmitter?  Seems like a real edge-case thing, but maybe I'm out of the loop.

shellixyz, 2019-03-10:
You can if you have a radio with OpenTX. You can't do that on all the radios though.

Teraformed, 2019-03-10:
Unfortunately my radio doesn't support that kind of thing in the Mixer. Btw It's a Flysky fs i6 with 14 ch mod.

> BTW only roll and pitch are inverted. Yaw stays the same.

correct.

teckel12, 2019-03-11:
I haven't mastered flying forward, so I'll trust the experts.

DzikuVx, 2019-10-09:
#5122 this should do it

DzikuVx, 2019-10-10:
will be a part of Global Functions in INAV 2.3, closing