iNavFlight/inav #4441: FEATURE REQUEST (and a longshot one at that): RC Ground Vehicle Mixer?

IGotYourDoughnut, 2019-03-03:
## Current Behavior
Currently INAV is exclusive to things in the air.

## Desired Behavior
Ability to utilize INAV Configuration App to easily set up flight controller for stability control, GPS, OSD, Smart Audio Support, S.Bus, and other features commonly found on multirotors and fixed wings but to be implemented on an RC truck, car, or buggy. 

## Suggested Solution
Add a mixer drop down menu under configuration tab for RC Ground vehicles

## Who does this impact? Who is this for?
The main impact for this is a side project I have going on for myself. Recently, I have started converting guys at the local track over to FPV setups and have really thought that this could be a cool thing to set up for an RC car to be able to have the telemetry flexibility of what we use in airplanes and quadcopters as well as OSD. I also see this being a great way to introduce those who have interest in the lunar rovers and Mars rovers into the hobby as well. 
 
## Additional context
I plan on taking an old beat up 2 Speed RC Monster Truck I have that I have placed a flight controller and FPV System with OSD on. Currently, I am using the flight controller to only monitor battery voltage and have the OSD to say my name in the bottom of the screen. My receiver (4 channel FRsky X4R) is bypassing the flight controller all together and going to the ESC, Steering Servo, and Transmission Servo via PWM lines. My intention with this monster truck is to setup a long range receiver via s.bus to the flight controller and utilize the flight controller to have selectable stability modes (similar to horizon mode vs acro mode), control throttle and direction, add an arm switch, select gears, monitor voltage and current, cycle power to LEDs, have GPS connectivity, add beeper, and fully functional OSD. The equipment I plan on using is listed below. 

**Flight Controller:**
 Matek F411-WING Flight Controller

**Receiver:**
FrSky R9 Mini Long Range Telemetry Receiver

**ESC:** 
Readytosky RC 120A Brushless ESC 2-4S Sensorless Speed Controller 

**GPS:**
RDQ BN-880 Flight Control GPS Module Dual Module Compass With Cable

**Camera:**
Foxeer Predator Micro

**Video Transmitter:**
RDQ Mach 3 Video Transmitter 

**Transmitter:**
FrSky X-Lite with R9M Module

**Servos (one for steering, one for transmission):**
Turnigy™ TGY-210DMH

rb1205, 2019-03-03:
I can't think of what you can possibly stabilize on an RC car with steering and throttle/brakes alone?

About the automatic navigation, you can probably do it easily by taking a fixed wing mixer and customizing it a bit. Try putting the rudder and a bit of roll command on the steering servo, it should do the trick.

Not sure how you could make the transmission work with inav, probably it's better to keep that bypassed on the rx.

IGotYourDoughnut, 2019-03-03:
I would imagine the breaking would be handled fine in the ESC as long as you would use the PWM out protocol on the Flight Controller. I should say, I am using the word stabilization very loosely. I know Traxxas and Spectrum have something in their receivers that does this but I am not sure exactly how. 

As for the "stabilization," my assumption and very limited knowledge of how the firmware works was that you could set something similar like you would with a rudder, where if your tail end kicked out from lack of traction, similar to how your tail end on an airplane could kick out from the wind, the flight controller would compensate accordingly.

The transmission is actually probably the easiest I would imagine, but again, I am a complete firmware noob. That you could just set on a switch with a high and low, similar to how you would set up a buzzer or PWM relay. The servo right now is hooked up to a switch on my X-Lite that when the switch is low, it moves the servo horn to the far left, and when it is high it moves the horn to the far right.

Nihilistic, 2019-03-04:
I don't know about stabilization for an RC ground vehicle, but autonomous navigation capability with GPS could be nifty. RTH in case of signal loss, only back tracking on street paths already taken, rather than trying to take a bee line back to home. WP navigation so you could send your RC truck a few blocks away to be filled up with a Big Mac and fries. Too drunk to drive, but out of beer and not ready to quit the all night dance party in VRchat? Beer Traxxis Truck to the rescue! :p Of course, this would require trustworthy vendors on the other end, and a local police force with a sense of humor, and probably some luck to pull off. There are companies working on robot vehicle delivery systems, so I bet it can be done, but would basically be the same complexity as self-driving cars, only on a smaller scale. Probably outside the scope of iNAV and the controller boards we're using now.

IGotYourDoughnut, 2019-03-04:
From the controller boards I am not quite convinced it is outside their realm of capabilities yet. I feel like the biggest problem will be obstruction control, but before I can get to playing with that I figured I would at least have my first deliverable as a truck that had GPS readout, telemetry, and steering stabilization to assist landing jumps that your rear end kicked out on. Even if all of this is only monitored in real time via an OSD that is a good enough base point as well. Low voltage beeper / LEDs is easy enough to get to work. Started wiring up a new test rig today for it. 

Really I think the area that would help actually make this work the most is adding  graphical mixer like we use for fixed wing / multirotor / tricopter for a ground RC. The biggest reason I came here over Betaflight was because I am betting a lot of tools to make this work are already in INAV and just haven't been applied/tried in land vehicles yet.

DzikuVx, 2019-03-04:
It is possible (theorethicaly) to configure rover with ESC braking. I never tested since I have zero rovers that are GPS capable.
What can be done right now:
1. Use AIRPLANE vehicle type
2. Set motor output like always
3. set one servo with Stabilized Yaw innput
4. enable 3D Mode (that will enable braking by motor)

And that's all. INAV will not reverse yaw control when throttle is below min and that's all

closed 2020-04-23 by DzikuVx