iNavFlight/inav #4443: UK Units

MartinHugh, 2019-03-03:
In the UK aviation, 

height is measured in feet, 
distance in metres/km, and 
speed in mph.  

I know, it is weird - but that is how it is.   I raised an issue a couple of years back which got looked at and closed.  But in the latest versions of iNav configurator I can not see a way of doing this.  

In any case perhaps the best of way of tackling this is to allow the user to set whatever units they wish for each metric.
 
See
https://github.com/iNavFlight/inav/issues/1853

thanks.

shellixyz, 2019-03-03:
You can select "UK units" in the OSD tab. The altitude uses metric units though. It will be fixed in the next release.

[image]

MartinHugh, 2019-03-03:
shellixyz. Many thanks.

wx4cb, 2019-03-03:
just as an aside - those of us old enough still use feet/miles for distance :D even in aviation :D none of this european stuff :D

shellixyz, 2019-03-08:
Fix merged in the development branch, closing

PR #4445 (2019-03-08): Use imperial units for altitude for UK
Fixes #4443
commit 8bcf368c (2019-03-08): Use imperial units for altitude for UK (#4445)

Fixes #4443
closed 2019-03-08 by shellixyz