iNavFlight/inav #4444: RTH does not work. Quad drifts away.

Teraformed, 2019-03-03:
## Current Behavior
The Quad drifts away when activating RTH.

## Steps to Reproduce
1. Take off with GPS Fix at 20 satellites. Barometer = OK, Magnetometer= OK. When armed "Distance to Home" at 0 meters as it should.
2. Hover about 10 meters infront of you at an altitude of ~ 5 meters.
3. Activate RTH
4. See your quad drifting away by circling out.

## Expected behavior
Quad lands where it was armed.

## Additional context
"RTH" is the reason why I flashed INAV on this board in the first place. Thats a bummer if it doesn't work correctly, because Betaflight's  "GPS Rescue" is working without problems though.

Unfortunately do not have Blackbox logs yet because forgot microsd card home. 

- FC Board name and vendor: Matek F722-SE
-Target: inav_2.1.0_MATEKF722SE.hex
- INAV version string: 2.1.4

[DUMP !]

# dump

# version
# INAV/MATEKF722SE 2.1.0 Dec  9 2018 / 16:40:40 (3dc3158ac)
# GCC-7.1.0

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
[... 764 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 1500
set battery_capacity_warning = 350
set battery_capacity_critical = 200
set battery_capacity_unit = MAH

shellixyz, 2019-03-03:
Please reproduce with an SD card inserted and provide a BB log. It is difficult if not impossible to know what happened without one.

DzikuVx, 2019-03-04:
The most probable cause is wrongly setup magnetometer. Either wrong orientation, bad calibration or hardware problems

Teraformed, 2019-03-05:
> Please reproduce with an SD card inserted and provide a BB log. It is difficult if not impossible to know what happened without one.

WIll do as soon as possible. Unfortunately it has been raining all day long. 👎

malckris, 2019-03-05:
Had a similar issue... what DzikuVx said, magnetometer might be setup incorrectly.. mine is setup like this "set align_mag = CW270FLIP" 
with the screen on Configurator that shows the quad orientation, tilt the nose up about 60 degrees, if it starts to rotate then change the magnetometer alignment until it stays put...

teckel12, 2019-03-05:
1) Make **SURE** compass alignment is correct
2) Calibrate compass outside away from interference
3) Test fly

Too many people skip step 1, or have no idea how to verify compass alignment is correct.

Teraformed, 2019-03-05:
> Had a similar issue... what DzikuVx said, magnetometer might be setup incorrectly.. mine is setup like this "set align_mag = CW270FLIP"
> with the screen on Configurator that shows the quad [...]

I just realized that I read this somewhere in the manual of GPS.

So I looked up and found this: 

INAV/BF: CW 270° Flip
Ardupilot/Mission Planner: Rotation None

might try those settings first. Pardon me.

teckel12, 2019-03-05:
@Teraformed There's only 8 orientations to try. Only one will work, which can easily be tested on the bench in seconds. If you're using a BN-880 ist probably 270 Flip or 90 Flip depending if you face it forwards or backwards. But still, takes a couple minutes at most to test all 8 orientations even if you're not sure at all.

But... Make SURE orientation is correct before you fly! And calibrate as well in an open area. Only then should you fly. A toilet bowl due to incorrect orientation can easily lead to a crash or hitting someone.

Teraformed, 2019-03-06:
> Had a similar issue... what DzikuVx said, magnetometer might be setup incorrectly.. mine is setup like this "set align_mag = CW270FLIP"
> with the screen on Configurator that shows the quad [...]

That did the trick. Now it does work even better than expected. 👍

closed 2019-03-06 by giacomo892