iNavFlight/inav #4451: Drone flip and crash

darkoudeas, 2019-03-03:
Dear all,

I understand this is not the proper place for such a question. But it is driving me crazy since it happened back in August and didn't manage to get help in forums. 

Omnibus F4 v5 equipped drone was flying just fine with Inav 1.9.1 . With no hardware changes I upgraded to inav 2.0 . After a few flights (generally it was flying smoothly) the following happened:

- While hovering drone flipped upside down and crashed (angle mode, GPS had locked). Looking at log all motors seem to stop at the same time (maybe an ESC issue???)
- After crash, motors kept spinning in high rpm and stopped only after disarmed from R/C.

Attaching log file (have a look at 9 sec) and cli diff
I personally don't believe it is a bug, but some kind of hardware "fault" or misconfiguration. I downgraded to 1.9.1 and had some test flights this week (7 batteries) without any problem. But I don't feel confident to make long range flights .

[cli_diff.txt](https://github.com/iNavFlight/inav/files/2923637/cli_diff.txt)
[log_RC_VID_0008.TXT](https://github.com/iNavFlight/inav/files/2923638/log_RC_VID_0008.TXT)

I appreciate any help and sorry for posting here for support issues.
Thank you

DzikuVx, 2019-03-04:
[image]

What I see in  the log is that was a combination of following things:
1. Your MR seems to be hovering on very low throttle. Maybe you should calibrate ESCs
1. You do not use AIRMODE that means you loose stabilization of low throttle
1. in 9.520 you pull throttle to ZERO , Iterm gets zeroed and UAV looses stability
1. Then for a few times you push throttle high and then back to zero (10.926 12.051 and so on) 
1. and then UAV hits the ground 14.31
1. at 15.989 you anable AIRMODE without disarming and UAV tries to regain stability on the ground

I see no software problem, only hardware/user related

closed 2019-03-04 by DzikuVx

darkoudeas, 2019-03-04:
Thank you very much for your response.

I just recalibrated ESCs and will see how it goes. 

1) Is angle mode protecting MR from flipping?
2) The AUX that enables airmode, also enables alt hold (same range) . Not really sure if activated during crash, but it logged only the airmode. Is this normal?

If you have 2 minutes to spend, have a look at the flight crash video. I understand that this is not the scope of github tracker.

https://arkoudeascom-my.sharepoint.com/:v:/g/personal/dimosthenis_arkoudeas_com/EcOirOJXyd5MkkSSSFpiexgBsn9ygIMkt21cLaD4n77p8Q?e=xTGFBR

DzikuVx, 2019-03-04:
Angle mode is not the solution. What you need is permanently enable AirMode (as a feature) which prevents those kinds of things where throttle is low.
There is nothing new on the video that did not showed on the log itself. Evident lack of conrol on low throttle