iNavFlight/inav #4455: suggest add new baro sensor "SPL06-001"

sincoder, 2019-03-04:
Seems a good replace for MS5611 
Feature:
 
Pressure: 300 - 1200 hPa. 
Temperature: -40°C to 85 °C.
Pressure sensor precision: ± 0.006 hPa (or ±5 cm) (high precision mode).
Pressure sensor accuracy: ± 0.06 hPa (or ±50 cm) (non -linearity), ±1 hPa (or ±8 m) (absolute).
Temperature accuracy: ± 0.5°C.
Pressure temperature sensitivity: < 0.5Pa/K
Measurement time: Typical: 28 ms. Minimum: 3 ms.
Average current consumption: High precision: 60 µA, Low power: 3 µA, Standby: <1 µA.
Supply voltage: VDDIO: 1.2 – 3.6 V, VDD: 1.7 – 3.6 V.
Operating modes: C ommand (manual), Background (automatic), and Standby.
Calibration: Individually calibrated with coefficients for measurement correction.
FIFO: Stores latest 32 pressure or temperature measurements.
Interface: I2C and SPI (both with optional interrupt)
Package dimensions: 8-pin LGA, 2.0 mm x 2.5 mm x 0.95mm.
 
Product Description:
 
The SPL06-001 is a miniaturized Digital Barometric Air Pressure Sensor with a high accuracy and a low current consumption. 
The SPL06-001 is both a pressure- and a temperature sensor. 
The pressure sensor element is based on a capacitive sensing principle which guarantees a high precision during temperature changes. 
The small package makes the SPL06-001 ideal for mobile applications and wearable devices.
The SPL06-001 ‘s internal signal processor converts the output from the pressure and temperature sensor elements to 24-bit results. 
Each pressure sensor has been calibrated individually and contains calibration coefficients. 
The coefficients are used in the application to convert the measurement results to true pressure and temperature values.
The SPL06-001 has a FIFO that can store the latest 32 measurements. 
By using the FIFO, the host processor can remain in a sleep mode for a longer period of time between readouts. 
This can reduce the overall system power consumption.
Sensor measurements and calibration coefficients are available through the serial I2C or SPI interface.
 
Typical Applications:
 
Indoor Navigation (floor detection e.g. in shopping malls and parking garages)
Health and Sports (accurate elevation gain and vertical speed)
Outdoor Navigation (GPS start-up time and accuracy improvement, dead-reckoning e.g. in tunnels)
Weather Station ('Micro-weather' and local forecasts)

giacomo892, 2019-07-06:
Attaching the impossible to find datasheet for public use
[21ic下载_数字压力传感器_SPL06-001_datasheet.pdf](https://github.com/iNavFlight/inav/files/3364424/21ic._._SPL06-001_datasheet.pdf)

giacomo892, 2019-07-06:
Attaching the impossible to find datasheet for public use
[SPL06-001_datasheet.pdf](https://github.com/iNavFlight/inav/files/3364426/SPL06-001_datasheet.pdf)

shellixyz, 2019-08-25:
Driver merged, closing

PR #5028 (2019-08-25): Add SPL006-001 barometer driver
Closes #4455 

The SPL006-001 is a precise barometer pin-to-pin compatible with the BMP280.

Configurator PR: https://github.com/iNavFlight/inav-configurator/pull/823
closed 2019-08-25 by shellixyz