iNavFlight/inav #4457: [FEATURE REQUEST] Add Pololu Serial Protocol support for external servos and IO

wx4cb, 2019-03-04:
Currently you are limited to the amount of servo outputs that are provided on the board (which can easily be exhausted)

Pololu offer different types of servo controller that  can be controlled via a UART and USB in their maestro line of products. from a small form factor 6 device up to a 24 device board.

All their docs are here for such a simple protocol. I was just wondering if there was a way that the protocol could be implemented so that aux channels could be moved to this board for things like head trackers and so forth?

Just a thought as that would free up some servo resources on the main board. for example. my F35 is full and i want to add a PT for the camera but can't do that as i'm using all the current output channels for flight control servos and escs.

If a DEV wants to take it on, i will quite happily provide hardware. This is already used on ardupilot for their antenna tracker firmware in mission planner

Product Pages: https://www.pololu.com/docs/0J38
Product: SDK https://www.pololu.com/docs/0J41
Protocol Documentation: https://www.pololu.com/docs/0J38

shellixyz, 2019-03-04:
Note that you can use a PCA9685 I²C IO extender

wx4cb, 2019-03-04:
agreed, but this is just another option plus i have 4 of these on hand :P

closed 2023-10-27 by DzikuVx