iNavFlight/inav #4465: MSP v2 python library?

wx4cb, 2019-03-05:
not strictly an "issue"

but does anyone know where I may be able to procure a python library for MSP v2 ?

the only one's i've found only support MSP V1.

I'm trying to write some code to talk to the flight controller and would prefer NOT to use mavlink :D

maybe @stronnag might be able to help.

giacomo892, 2019-03-05:
Please use telegram, facebook or rcgroups for such requests. You can find all the references in the manual.

Closing

closed 2019-03-05 by giacomo892

wx4cb, 2019-03-05:
@giacomo892 there is nowhere in the wiki other than these two pages (which describe the frames and inave specific changes) that describe the MSP V2 protocol.

that's the issue.

https://github.com/iNavFlight/inav/wiki/MSP-Navigation-Messages
https://github.com/iNavFlight/inav/wiki/INAV-MSP-frames-changelog

giacomo892, 2019-03-06:
In open source projects there is a big rule that stands above everything else: 
_the source code is the best manual you can get_

fiam, 2019-03-07:
@wx4cb 
https://github.com/iNavFlight/inav/wiki/MSP-V2

wx4cb, 2019-03-08:
@fiam  thank you. that's all i was asking for :D

annoyinganime, 2020-04-24:
@wx4cb did you manage to make it work? I'm also trying to implement one, but for some reason it doesn't work

ricardodeazambuja, 2020-07-05:
@wx4cb and @annoyinganime, I developed a Python library that supports MSP V2 so you may want to have a look:
https://github.com/thecognifly/YAMSPy