iNavFlight/inav #4466: Gps ports assignment for Matek f722 SE

rcjarvis, 2019-03-06:
## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

rcjarvis, 2019-03-06:
Hi, Still experimenting with Inav.. My tricopter now flies like a dream !... But now I'm back to my quad and I received a GPS today and I want to install it on my f-450 ( yep old school crisis) and I installed the GPS as per Matek's recommendation under Inav wiring... But.. when I got in Inav I noticed port uart1 was set to MSP so I changed it to smp off and selecte sensor/gps baud 57600 as recommended by Matek...  No motor working... tried a few thigs nothing worked... SOOO ... I open up the top plate, installed the tx/rx gps wires to tx rx 3, set the sensor rolldown toGPS baud rate to 57600 and wolà.... for motors to work I just set back the uart1 to smp On and voilà... lots of fun can't wait to fly this quad...

closed 2019-03-06 by giacomo892

rolex24hrs, 2019-04-22:
Good morning rcjarvis. 
I have been having problems setting up my gps in inav (matek gps/compass and f722 std FC). It is soldered to uart1, set in inav to 57600, but I am still getting the red symbol at the top of the page. My compass and everything else is recognized, just not the gps. Any ideas? Is it because I am on uart1 while the usb is plugged in?

rcjarvis, 2019-04-22:
I got it working on uart 4 , if I remember correctly, uart1 is used for esc’s since I could not spin the motors when I soldered and assigned uart1 to the gps...

rolex24hrs, 2019-04-22:
Thanks for the reply. 
In my case my motors are spinning (I have my esc's wired into a seperate matek PDB), but I think I`m at loose ends, so I am going to try it your way just to see if things get better. I am cross eyed at looking at a hundred different posts trying to find a solution. I am assuming I just have to solder the rx and tx wires and can leave the 5v and G the same? The gps led's are lighting up like they should.

rcjarvis, 2019-04-23:
yes , do remember that the tx from the gps goes to the rx on the fc an rx of gps goes to tx on fc...
yes 5v and G should remain the same. from what your saying, you don't have a magnetometer on the gps? that would require 2 more wires (SDA and SCL) remember to set UART3 (I just checked and port 4 was used for gps on my tricopter because of the servo but on my quads it is port 3)  to gps under  sensors: