iNavFlight/inav #4469: Disarm when switching from Altitude Hold

malckris, 2019-03-07:
## Current Behavior

New hexacopter build, 3 previous flights for vibration and tuning, 4th flight was testing altitude and position hold.
switching altitude/position hold on/off several times worked perfectly.. final switching of altitude hold off crashed the aircraft.. craft was on its back motors still running, was unable to disarm, tried cycling arm switch.. finally had to disconnect battery.. looking at blackbox log it shows the disarm at the same instant I switched off altitude hold. 
Craft was in a 15 ft hover and I was about 30 ft away, very good signal strength at the time.  

Failsafe was set to land with 1000 throttle and 20 second delay... actually thought I had the throttle  setting at 1400... also it didn't really land it flew off at a steep decent..

Not sure why mode change would have caused a disarm, I did read through the similar issues that were shown and didn't see anything really related

I have tried switching nav modes while armed, but on the ground for obvious reasons.. I have not been able to duplicate yet..

TX: Horus X10 w/ OpenTX
RX: X4R with telemetry

## Suggested solution(s)

Possibly using a different FC?? I have an OMNINXT F7 on the way... I never felt 100% comfortable the the Kakute or really the lack of support for it..

## Additional context

[INAV_cli_680Hexa_KakuteF7- iNav_20190306_161454.txt.zip](https://github.com/iNavFlight/inav/files/2938753/INAV_cli_680Hexa_KakuteF7-.iNav_20190306_161454.txt.zip)
[LOG00004.TXT.zip](https://github.com/iNavFlight/inav/files/2938754/LOG00004.TXT.zip)

---

- FC Board name and vendor: Holybro Kakute F7
- INAV version string: INAV/KAKUTEF7 2.1.0 Feb 25 2019 / 16:48:16  (65b0ec142)

DzikuVx, 2019-03-08:
> Failsafe was set to land with 1000 throttle and 20 second delay... actually thought I had the throttle setting at 1400... also it didn't really land it flew off at a steep decent..

Set thrott;e is not land. It's more less falling down ith min throttle. If you want copter to land, use land.

I took a look at log andfrom the flight controller everything looks ok. At one point disarm came in

Check radio

stronnag, 2019-03-08:
> Check radio

From the log ...

```
$ inav_modes.rb /tmp/LOG00004.TXT 
LOG00004.TXT: INAV 2.1.0 (65b0ec142) KAKUTEF7
Iteration	Time(s)	Elapsed(s)	State
        0	  94.5	(   0.0)	nav_state_undefined (0)
    40544	 175.9	(  81.3)	nav_state_althold_in_progress (3)
    65216	 225.3	( 130.8)	nav_state_poshold_3d_in_progress (7)
    71936	 238.8	( 144.3)	nav_state_althold_in_progress (3)
    72000	 238.9	( 144.4)	nav_state_idle (1)
    81824	 258.7	( 164.1)	nav_state_althold_in_progress (3)
   101792	 298.7	( 204.2)	nav_state_poshold_3d_in_progress (7)
   124672	 344.6	( 250.0)	nav_state_althold_in_progress (3)
   124736	 344.7	( 250.2)	nav_state_idle (1)
   135904	 367.1	( 272.6)	nav_state_althold_in_progress (3)
   137455	 370.2	( 275.7)	end of log
Disarmed by: SWITCH
```

malckris, 2019-03-08:
I was thinking that somehow a switch disarm was accidentally sent, definitely wasn't bumped due to location of switch... after the disarm happened it was a fly-away and motors running until I pulled battery, shouldn't a true disarm have killed all power to the motors?... craft was on its back and the motors weren't really working all that hard, motors are 5210 340KV on 6s battery, if they had any reasonable power going to them I think they would have destroyed the props (only had minor tip damage on one prop) also not sure if all the motors were running... looking back at the log, all activity stopped logging right at the instant of the disarm flag, I've done several tests and the activity stops about 0.3 seconds later... I'm leaning towards a FC glitch, but wanted to report just in case it might be code...

malckris, 2019-03-08:
> Set thrott;e is not land. It's more less falling down ith min throttle. If you want copter to land, use land.
>
> I took a look at log andfrom the flight controller everything looks ok. At one point [...]

normally when I'm testing I set the throttle for a 'soft' fall in case things go wrong and I need to use the failsafe switch.... which in this case the failsafe and kill switch had no response..

Jetrell, 2019-03-08:
>
> 
> Set thrott;e is not land. It's more less falling down with min throttle. If you want copter to land, use land.
> 

With all respect. Can you explain this Pawel....

Have you or any of the Dev's ever tested the so called LAND function, that is supposed to be part of the failsafe **RTH procedure?**
[image]

I'm not referring to the failsafe **LAND procedure** that uses a set throttle and timer. It does work if selected. 

I'm referring to the way that the quad should land in **RTH procedure** by barometer and IMU, if the GPS signal is lost.

(From the failsafe wiki)
_* If any required navigation sensor becomes unavailable during a Return to Home (eg. loss of GPS fix), an emergency landing similar to SET-THR but with barometer support will be performed after a short timeout. An emergency landing would also be performed right when the failsafe is triggered if any required sensor is reported as unavailable._

I have tested this SO called function many times. And all it does is drop the throttle down to zero. And the quad falls to its death. It does not revert to the LAND procedure Throttle value, nor does it work with the Timer under that procedure. 
Why is this the case??

I truly don't understand why it can not land the quad by the barometer and IMU, the same way that a RTH auto land would occur. And then shut the motors off. Like it does under the RTH procedure with GPS lock.
I realize that it would not be able to hold position without GPS. But at least it would still land under power and shut the motors off after touch down.
Instead it coming crashing to the ground. Then runs your battery flat if it has no RF link, because the motors keep spinning in AIR mode.(even at zero throttle). 

Please reply. This is a bad bug that has been here for many releases. And should be address for safety reasons.

malckris, 2019-03-10:
just to update.... thinking it might be a bad FC... had another flyaway yesterday, logs showed one motor at 0%, one motor at 3%, 2 motors at 19% and 2 motors working normally... only thing was I had it in a hover for several seconds before it crashed... the flyaway acted as the logs were saying, but never showed lift off and hover... I did another successful flight afterwards for 17 minutes, that flight in the beginning logged the same motor behavior, which I believe was correct since I was on a slope and once armed and the throttle initiated the craft self leveled with the front right skid still touching the ground, but then showed power applied and the flight as expected... going through the second log there was 2 spots where the recording stopped, each less than a second (not sure if that means anything)...and on the second flight I did change the gyro LPF cutoff  from 256Hz to 98Hz, is there any chance the higher rate might have caused the problem? 
I'm very new to this, very steep learning curve....      

thanks in advance....
[FLYAWAY_LOG00008.TXT.zip](https://github.com/iNavFlight/inav/files/2950192/FLYAWAY_LOG00008.TXT.zip)
[NORMAL_LOG00009.TXT.zip](https://github.com/iNavFlight/inav/files/2950193/NORMAL_LOG00009.TXT.zip)

malckris, 2019-03-22:
I replaced the Kakute F7 with an OmniNXT F7 (wanted LED lights which the Kakute doesn't support)... I also got another Kakute F7 that I put in a 250 hex frame and configured it identical to the 690 hex that was having the issues... Both are not having any issues, and I have extensively tested the Kakute.. 
Really think it was a bad FC...

closed 2019-03-22 by malckris