iNavFlight/inav #4470: Matek 405 ctr issue > inav 2.0.0 and beyond

Gabbocombinaguai, 2019-03-07:
Hello everyone.
I set inav 1.9.1 on my matek 405 ctr and everything is fine, but when i try to upgrade to inav 2.0 or 2.1 bad happen. In the Pre-arming checks everything is green but the motors do not spin at all.

More over when i use inav 2.0 and 2.1 on the fc there is also a bug, i cant see the motor tab so i cannot calibrate mtors.

Here the dump

https://pastebin.com/kNn11dm7

I have no idea how to fix this.
Could you please give me a tip?

teckel12, 2019-03-07:
@Gabbocombinaguai Are you using the latest configurator? Are you doing a full wipe when flashing?

Gabbocombinaguai, 2019-03-07:
When I use 1.9.1 configurator and 1.9.1 fw everything is fine, but if i try to use configurator 2.0 and beyond i get stuck.
Always full chip erase when flashing.

teckel12, 2019-03-07:
@Gabbocombinaguai There were some issues with some early 2.0 configurators.  I would suggest using the latest configurator and INAV and doing a full chip erase.  My guess is that you were running a slightly older configurator with the latest INAV, which could yield strange results.

Gabbocombinaguai, 2019-03-07:
I have just done what you said to me. Using conf 2.1.4 and installing fw 2.1.0 ( stable version + full chip erase) I have bugs. No motors shown inside the motor tab, I cannot  even use motors using the scrolling bar in the motor bar ( i can scroll but no action follows) and althought arming is able no spinnig :/

teckel12, 2019-03-07:
@Gabbocombinaguai Go into the mixer tab and set that first. Also, see the upgrade documentation for INAV 2.0 (I believe this is what you missed).

Gabbocombinaguai, 2019-03-07:
Mixer? :) I've never used that. i will try . Thx

teckel12, 2019-03-07:
@Gabbocombinaguai This changed with 2.0.  If upgrading from 1.9 you need to set your mixer when you didn't need to previously.

Gabbocombinaguai, 2019-03-07:
Thx a lot. Problem solved 👍

closed 2019-03-07 by Gabbocombinaguai